micropython/micropython #2125: Add support for running Python code in the VM with pending exception check disabled

pfalcon, 2016-05-28:
Based on the discussion in https://github.com/micropython/micropython/commit/4681b86850076c013be2477d3850fd536361b62e#commitcomment-17591565

> I agree. When a non-main task is running in the VM, pending exceptions should be "disabled" (ie not processed) just like an irq mask for hard interrupts.

pfalcon, 2016-05-28:
An easy way to achieve this would be:

1. Disable pending exception check in jump opcodes.
2. Expose vm_hook_divisor externally.
3. Set it to MAX_INT when there's a need to run code with "pending exceptions disabled". Reset to normal value after such code has run.

dpgeorge, 2017-06-14:
This is now supported: with MICROPY_ENABLE_SCHEDULER enabled calling mp_sched_lock() will prevent the VM from handling any pending exceptions (or executing any scheduled callbacks) until mp_sched_unlock() is called.

closed 2017-06-14 by dpgeorge