micropython/micropython #2126: Documentation: there's no math module on the WiPy

marcelstoer, 2016-05-29:
`/docs/wipy/general.rst` clearly says 

> Due to space reasons, there’s no floating point support, and no math module.

Yet, https://micropython.org/resources/docs/en/latest/wipy/library/math.html gives the impression that the WiPy would support the math module. Primarily that's because the URL path is /**wipy**/library/math.html.

pfalcon, 2016-05-29:
How did you arrive at that URL?

marcelstoer, 2016-05-29:
- https://micropython.org/resources/docs/en/latest/wipy/
- search for 'math'
- click the first search result

vmesel, 2016-05-29:
@marcelstoer This is only explaining the normal math function on micropython, there isnt any place that says its specifically for the WiPy!

marcelstoer, 2016-05-30:
>  there isnt any place that says its specifically for the WiPy!

I know but it also doesn't say it be specifically NOT for the WiPy. The path /wipy/library/math.html makes it look like it were supported on the WiPy - contrary what it says elsewhere. So, there should either be a big fat warning at the top of this page or the path should be invalid.

vmesel, 2016-05-30:
Thats actually a mistake in the docs. If some contributor see this should help us

dpgeorge, 2016-05-30:
@danicampora what do you suggest to do?

danicampora, 2016-05-30:
I believe the problem is the way we are generating the docs. The directive:

```
.. only:: port_wipy
```

doesn't fully work in the sense that still all files in the directory are generated and visible via the search function. One way out would be to manually copy the right files into a separate folder, and generate from there. @dpgeorge: any other ideas?

pfalcon, 2016-05-30:
> doesn't fully work

So, the problem is not how we generate the docs, but how sphinx works. Fixes as deep as that thus would rather be considered on its side.

@vmesel perhaps randomly, but shoot straight in the middle of it:

> This is only explaining the normal math function on micropython

There will be more of that - MicroPython's docs should describe *MicroPython*, not a particular port. It's pending on my task list to add a corresponding notice to the Library Reference page, along the lines of "this described general MicroPython libraries, any module/function/class may be absent in a particular port, please refer to "General information" section of a port regarding what's available in it". Particularly in this case brought up by @marcelstoer , "General information" prevails (it's just not described in the docs so far).

Of course, it would be nice to make it easier for users of the officially supported (i.e available in mainline) ports, and we know how to do it: "Availability:" line, the same as CPython marks Linux/MacOS/Windows differences. We definitely should do that for math module too, as that's teh best resolution to this ticket, but it doesn't scale at all. As in: I don't see someone go over all modules/functions/classes and marking what's available in which port. We don't have enough people to document all what's available to start with.

marcelstoer, 2016-05-30:
> MicroPython's docs should describe *MicroPython*, not a particular port.

I fully agree. It really surprised me to see both the WiPy and the ESP8266 code in this repository. No wonder the documentation doesn't clearly separate between the language in its ports either.

> I don't see someone go over all modules/functions/classes and marking what's available in which port.

Not _someone_ but a _group of_ someones. Such tasks become much less daunting if you share the load. I know what I'm talking about. Last December/January I ported (i.e. re-wrote) the entire NodeMCU firmware documentation to MkDocs/RTD with a few fellow committers. We played divide & conquer and worked our way through some 40 modules like that. 

> "Availability:" line, the same as CPython marks Linux/MacOS/Windows differences

From a user's perspective I don't find that desirable at all. If I'm writing code for the WiPy I'd like to have docs for the WiPy. I'd find it irritating to get whole pages or paragraphs marked as "pyboard only" or so.
Is it really that you need to differentiate between the various ports at the function level (rather than module level)? If not you could simply build individual sets of documentation by hand-coding the TOC for each thereby only "including" the library pages actually available for a specific port.
I realize I don't know enough about your project yet to make bold claims but in my naive understanding of the structure that'd might be an alternative.

pfalcon, 2016-06-19:
> I realize I don't know enough about your project yet to make bold claims but in my naive understanding of the structure that'd might be an alternative.

Indeed. If you'd look at it, you'd find that all thing you talk about are (of course!) how it's done. That's of course good sign, meaning that we're on the same wave, so feel free to contribute fixes.

In the meantime, as expected, that was a Sphinx issue. There's a long history of community outcry regarding how Sphinx works in that respect: https://github.com/sphinx-doc/sphinx/issues/2150 . So the obvious solution was to solve a last mile problem with it and put together easily usable extension to workaround those issues (like from the issue above).

Fixed.

closed 2016-06-19 by pfalcon