micropython/micropython #2129: Can't compile firmware for esp8266

kommancs96, 2016-05-30:
I was trying to compile the firmware for esp8266 chip.
I followed the steps required, and successfully installed the Esp open sdk on my computer. However I get 
some errors when trying to compile the firmware. 
After running the `make` command I get the following output:
```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/qstr.i.last
In file included from ../lib/axtls/ssl/ssl.h:75:0,
                 from ../py/../extmod/modussl.c:37:
../lib/axtls/ssl/tls1.h:43:21: fatal error: version.h: No such file or directory
 #include "version.h"
                     ^
compilation terminated.
make: *** [build/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build/genhdr/qstr.i.last'
```
I am using Mac OS X 10.11.5

bsdfox, 2016-05-31:
did you skip the 'make axtls' step?

vmesel, 2016-06-01:
Do you have the ESP toolchain for compiling the firmware?

pfalcon, 2016-06-12:
@kommancs96 : There're currently no know issues with building for esp8266. Please be sure to follow the instructions exactly. There're additional instructions at http://forum.micropython.org/viewtopic.php?f=16&t=1689 . That topic also contains requirements for bug reporting via this tracker.

closed 2016-06-12 by pfalcon