micropython/micropython #2130: timezone support for ntptime

puuu, 2016-05-30:
From the release notes of MicroPython ESP8266 firmware v1.8 is clear, `ntptime` only supports UTC. I agree, time zone handling is out of the focus of `ntptime`. But it would be nice, if `ntptime.settime()` would support a offset argument, which is be added to `ntptime.time()` before setting the rtc.

Something like `ntptime.settime(offset=2*3600)` to provide CEST.
Then, getting the local time is easy as `time.localtime()` (if #2097 is solved).

I believe, this is a simple api enhancement. Later, a timezone module may calculate the offset by time zone and daylight saving.

I can provide a pull request, but I am not sure about the api and what unit (seconds, minutes or hours) is advisable for the offset. Also the PR may conflict with #2121 .

pfalcon, 2016-05-30:
timezone support in `ntptime` depends on timezone support in `time`.

puuu, 2016-06-01:
@pfalcon, thanks for pointing out timezone support in `time`. I had a look to the CPython time module. There I found, the timezone offset (in seconds) is in `time.timezone`.
@dpgeorge is there any reason, why we can't have a module variable `time.timezone`? Then the value can be added to `seconds` in `time.localtime()`. Unfortunately, I don't know how to add a module variable which I can be accessed in a C function.

IMHO, this would be a simple timezone support.

pfalcon, 2016-06-01:
Right, time.timezone is the Python way to do it. Unfortunately, MicroPython doesn't support assignment to C-level module variables in a general case. We have only adhoc support for overriding stuff in "builtins" module, and I use to make a patch for adhoc overriding of sys.stdout, which is still not in the mainline. So, this question needs additional consideration and thus some time.

puuu, 2016-06-01:
Thank you for clarifying. Should we leave this issue open, as general timezone feature request?

pfalcon, 2016-06-01:
> Should we leave this issue open, as general timezone feature request?

Yup, with a quick subject edit it becomes such ;-).

dpgeorge, 2016-06-28:
Note that writing to time.timezone does not seem to change the timezone.  The time.timezone variable is just a mirror of the underlying C value, which is set when tzset() is called.  tzset() cannot be used to change the timezone, only to (re)initialise it from the environment variable 'TZ'.  There does not seem to be a way to change the timezone in a purely Python way.  The only way to do it is call the OS or set the environment variable 'TZ' to something different.

zed, 2016-07-01:
`time.timezone` is not the Python way to do it. See [Getting computer's UTC offset in Python](http://stackoverflow.com/q/3168096/4279). If the system supports it then `time.localtime().tm_gmtoff` is the way to get the current utc offset for the local time zone. If it doesn't then a portable way to get the utc offset including the support for past/future dates (when the utc offset may be different) is to use `pytz` timezones:

```
from datetime import datetime
from tzlocal import get_localzone # $ pip install tzlocal

tz = get_localzone() # local timezone 
d = datetime.now(tz) # or some other local date 
utc_offset = d.utcoffset().total_seconds()
```

It works for almost all timezones from [the tz database](https://en.wikipedia.org/wiki/Tz_database).

puuu, 2016-07-27:
@dpgeorge Thought, setting and using `time.timezone`, `time.altzone` and `time.daylight` is an easy and lightweight way to implement basic timezone support in micropython. Since there does not seem to be a way to change the timezone in a purely Python way and on micro-controller, micropython is its own OS, setting this variables IMHO would be the most straight forward way.

Implementing a timezone module with a timezone database could be  an additional task, using this interface. IMHO most micropython project are fine by settings this variables.

pfalcon, 2017-06-11:
Let's breath new life into this with https://github.com/micropython/micropython/issues/3141

dpgeorge, 2017-06-13:
time.timezone is essentially read-only in Python.  Setting the timezone should be done naturally using machine.RTC.

carstenblt, 2017-12-25:
The esp-idf for the ESP32 supports timezones, right now I have implemented a machine.RTC().settimezone() method to use it. Based on the [deepsleep code](https://github.com/micropython/micropython-esp32/pull/85) by MrSurly that's still pending. It works nicely with ntptime.

In case that's something you'd like to see, I'd be happy to submit the (really simple) code. But I have the feeling you're looking for a more general, platform independent approach?

pfalcon, 2018-09-16:
> time.timezone is essentially read-only in Python.  Setting the timezone should be done naturally using machine.RTC.

Answering this old comment, time.timezone is essentially read-only in *CPython*. Just like many other warts and accidental gaps it has. There's no reason why CPython's disabilities should be mirrored in MicroPython. Using time.timezone value (reading it) is the obvious way to provide timezone support, consistent across ports. Then allowing to *write* it is the obvious way to let users adjust it. Then the only port/platform specific bit is setting initial value for (u)time.timezone. And indeed, assuming it's done right, users can treat (u)time.timezone in a CPython-like "read only" way. And if it's not done right, it's trivial to adjust it.

I'm writing (u)time.timezone, because implementing time.timezone variable would be trivial for Pythin "time" module, sitting on top of "utime" module. But so far, Python stdlib modules aren't really shipped (much) for bare-metal ports, so doing it on "utime" level while more complex, would allow to truly system-wide solution.

Schopenhuaer, 2021-02-25:
def set_utctime(offset=0):
    do_connect()
    from machine import RTC
    rtc = RTC()
    import ntptime
    ntptime.settime()
    date_time = list(rtc.datetime())
    date_time[4] = date_time[4] + offset
    date_time = tuple(date_time)
    rtc.datetime(date_time)
    print(rtc.datetime())

def do_connect():
    import network
    sta_if = network.WLAN(network.STA_IF)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.active(True)
        sta_if.connect('your ssid', 'your wifi password')
        while not sta_if.isconnected():
            pass

Schopenhuaer, 2021-02-25:
from machine import RTC
rtc = RTC()
import ntptime
ntptime.settime()
date_time = list(rtc.datetime())
date_time[4] = date_time[4] + offset
date_time = tuple(date_time)
rtc.datetime(date_time)

peterhinch, 2021-02-25:
Does not handle day rollover if called close to 00:00hrs. A solution has been discussed in [the forum](http://forum.micropython.org/index.php).