micropython/micropython #2132: Periodic messages like: chg_A3:-180

robert-hh, 2016-05-30:
With the latest build for esp8266, after reset I get periodic messages alternating "chg_A3:-180" and "chg_A3:0" about every 16 seconds. The series stops after 8 messages (4 pairs).

MicroPython v1.8-159-g7d4fd8b-dirty on 2016-05-30; ESP module with ESP8266

dpgeorge, 2016-05-30:
This is the same as issue #2005.

closed 2016-05-30 by dpgeorge

robert-hh, 2016-05-30:
Sorry. I though I had seen that, but could not find it again.

pfalcon, 2016-05-30:
@robert-hh : that message has nothing to do with the latest build or anything on MicroPython side. Most likely, it's related to varying WiFi signal conditions. On one of my board, I get such messages very rarely. On another, regularly. So, ignore or disable ESP RTOS logging as described in the linked issue.

robert-hh, 2016-05-30:
Thanks Paul, understood. It's only that I've seen it yesterday the first
time and assumed, it was related to the most recent changes. But maybe
that was related to the use of the most recent open-esp-sdk, which I
installed over the weekend.