micropython/micropython #2134: How can I Start Contributing to the Project?

vmesel, 2016-05-31:
Hey @dpgeorge, I'm Vinicius and I would like to start contributing to MicroPython as an opensource Python developer and as a Documentation Translator to Portuguese.

Firstly, I have to ask, what you think needs to be done so I can start developing on the core of the MicroPython software?

And the last question, What I could do as a member of the Brazilian Python Association to start encourage people to use MicroPython rather than Arduino or other existing softwares.

vmesel, 2016-06-01:
Already got some ideas, closing the issue!

closed 2016-06-01 by vmesel

pfalcon, 2016-06-02:
Great work! Please make sure you don't miss  http://forum.micropython.org/viewtopic.php?f=3&t=779 ,  https://github.com/micropython/micropython/issues/1331 , https://github.com/micropython/micropython/wiki/ContributorGuidelines