micropython/micropython #2138: esp8266: rtc.memory() crash if rtc memory is uninitialised (i.e. at power on)

MarkR42, 2016-05-31:
Steps to reproduce:

1. power off device

2. Power on and enter at REPL:
```

>>> import machine
>>> rtc = machine.RTC()
>>> rtc.memory()

```

Expected result:

Returns None, at least, looking at the source it looks like it should.

Actual result:

Crash:

```
import machine
>>> rtc = machine.RTC()
>>> rtc.memory()

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 31628, room 16 
tail 12
chksum 0x53
ho 0 tail 12 room 4
load 0x3ffe8000, len 1040, room 12 
tail 4
chksum 0x9f
load 0x3ffe8410, len 3112, room 4 
tail 4
chksum 0xe8
csum 0xe8

```

MarkR42, 2016-05-31:
NB: works normally if rtc memory is initialised:
```

>>> import machine
>>> rtc = machine.RTC()
>>> rtc.memory(b'hello')
>>> rtc.memory()
b'hello'
```

And after reset, retains rtc memory as expected.

dpgeorge, 2016-06-01:
Fixed.  See commit above.

closed 2016-06-01 by dpgeorge