micropython/micropython #2139: Instance.__dict__ is read-only

prusnak, 2016-06-01:
Micropython:

```
>>> class Foo:
...     pass
>>> t = Foo()
>>> t.__dict__
{}
>>> t.__dict__['a'] = 4
>>> t.__dict__
{}
>>> t.a
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Foo' object has no attribute 'a'
```

Python3:

```
>>> class Foo:
...     pass
... 
>>> t = Foo()
>>> t.__dict__
{}
>>> t.__dict__['a'] = 4
>>> t.__dict__
{'a': 4}
>>> t.a
4
```

Same applies for t.__dict__.update(another_dict)

prusnak, 2016-06-01:
As per https://github.com/micropython/micropython/blob/master/py/objtype.c#L475 this is expected. Feel free to close if there is no plan to address this issue.

dpgeorge, 2016-06-01:
Please see original discussion in #1757 where `__dict__` was first introduced.

pfalcon, 2016-06-02:
@prusnak: Feel free to add this to https://github.com/micropython/micropython/wiki/Differences , otherwise yes, this would be closed as "by design".

prusnak, 2016-06-03:
Added as point #9 in https://github.com/micropython/micropython/wiki/Differences#implementation-differences and closing. Thanks for the explanation!

closed 2016-06-03 by prusnak