micropython/micropython #2141: WiFi connect retrial spamming Serial Port

vmesel, 2016-06-01:
Hey guys, I'm really trying to use my ESP8266 but I'm getting this error/info on the serial usb.
Is there any way to stop this popping error?

```
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (19)
reconnect
f -180, scandone
no GVT-C05C found, reconnect after 1s
reconnect
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 
```

vmesel, 2016-06-01:
Just reset your ESP8266 and it will be running well!

pfalcon, 2016-06-02:
> Is there any way to stop this popping error?

The usual one - reading the docs: http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html#general-board-control , searching previous issues: https://github.com/micropython/micropython/issues/2005, https://github.com/micropython/micropython/issues/2072

closed 2016-06-30 by pfalcon