micropython/micropython #2142: Clear the MicroPython Screen

vmesel, 2016-06-02:
There is already a code for cleaning the screen, but it's not available natively in MicroPython. How about implementing it?

```
class clear():
    print("\x1B\x5B2J", end="")
    print("\x1B\x5BH", end="")
```

peterhinch, 2016-06-02:
Users connect to boards running MicroPython with assorted terminal programs and with rshell. Does the above code assume a particular terminal emulation?

vmesel, 2016-06-02:
Nope! like when I'm in my MacBook pro or in Windows, I use respectively, Terminal and PuTTY and both of them need some kind of code to clean the terminal!

dpgeorge, 2016-06-02:
It could be part of upysh (see eg https://github.com/micropython/micropython-lib/pull/75).

vmesel, 2016-06-04:
I'm going to make a FORK of the micropython-lib so I can work on this and than I will make a Pull Request!

dpgeorge, 2016-06-05:
It looks like someone beat you to it: https://github.com/micropython/micropython-lib/pull/76

vmesel, 2016-06-05:
LOL... Thats ok! It's for the community!

closed 2016-06-05 by vmesel

Jibun-no-Kage, 2024-10-08:
The above codes are just classic ascii escape sequence codes by the way. 2J is clear screen, H is return cursor to home.  Need both since 2J does not change the cursor position.