micropython/micropython #2143: Error finding endian.h during make deplibs Mac OS 10.11.5

Cabalist, 2016-06-02:
Hello

I am building micropython from source on Mac OS 10.11.5.  I have libffi installed with brew.  I have tried building this with ```PKG_CONFIG_PATH``` set to the path to libffi and not.   I can see ```endian.h``` in /usr/include/machine/ but it looks like the build script cannot?

```
Works-MBP:unix work$ make clean
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Package libffi was not found in the pkg-config search path.
Perhaps you should add the directory containing `libffi.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libffi' found
Package libffi was not found in the pkg-config search path.
Perhaps you should add the directory containing `libffi.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libffi' found
rm -f micropython
rm -f micropython.map
rm -rf build

Works-MBP:unix work$ make deplibs
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Package libffi was not found in the pkg-config search path.
Perhaps you should add the directory containing `libffi.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libffi' found
Package libffi was not found in the pkg-config search path.
Perhaps you should add the directory containing `libffi.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libffi' found
cd ../lib/libffi; git clean -d -x -f
Removing Makefile.in
Removing aclocal.m4
Removing autom4te.cache/
Removing build_dir/
Removing compile
Removing config.guess
Removing config.sub
Removing configure
Removing depcomp
Removing fficonfig.h.in
Removing include/Makefile.in
Removing install-sh
Removing ltmain.sh
Removing m4/libtool.m4
Removing m4/ltoptions.m4
Removing m4/ltsugar.m4
Removing m4/ltversion.m4
Removing m4/lt~obsolete.m4
Removing man/Makefile.in
Removing mdate-sh
Removing missing
Removing testsuite/Makefile.in
cd ../lib/libffi; ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: glibtoolize --copy --force
glibtoolize: putting auxiliary files in '.'.
glibtoolize: copying file './ltmain.sh'
glibtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
glibtoolize: copying file 'm4/libtool.m4'
glibtoolize: copying file 'm4/ltoptions.m4'
glibtoolize: copying file 'm4/ltsugar.m4'
glibtoolize: copying file 'm4/ltversion.m4'
glibtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: running: /usr/local/Cellar/autoconf/2.69/bin/autoconf --force
autoreconf: running: /usr/local/Cellar/autoconf/2.69/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:31: installing './compile'
[... 357 lines cut ...]
         ^
1 error generated.
In file included from sha256.c:32:
In file included from ../ssl/os_port.h:40:
../crypto/os_int.h:60:10: fatal error: 'endian.h' file not found
#include <endian.h>
         ^
1 error generated.
In file included from sha384.c:32:
In file included from ../ssl/os_port.h:40:
../crypto/os_int.h:60:10: fatal error: 'endian.h' file not found
#include <endian.h>
         ^
1 error generated.
In file included from sha512.c:32:
In file included from ../ssl/os_port.h:40:
../crypto/os_int.h:60:10: fatal error: 'endian.h' file not found
#include <endian.h>
         ^
1 error generated.
make[2]: *** No rule to make target `.depend', needed by `all'.  Stop.
make[1]: *** [target] Error 2
make: *** [axtls] Error 2
Works-MBP:unix work$ 
```

Anton-2, 2016-06-02:
Indeed, make deplibs does not work currently on macOS. However you don't need it if you just want to compile micropython from source.
Make sure that PKG_CONFIG_PATH points to the directory where `libffi.pc` is. On your pasted exemple it's not.
On my system `PKG_CONFIG_PATH=/usr/local/Cellar/libffi/3.0.13/lib/pkgconfig make` works, you may need too adjust the path for the libffi version, or homebrew root dir.

Cabalist, 2016-06-02:
Same error :/  
```
Works-MBP:unix work$ PKG_CONFIG_PATH=/usr/local/Cellar/libffi/3.0.13/lib/pkgconfig make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/qstr.i.last
In file included from ../py/../extmod/modussl.c:37:
In file included from ../lib/axtls/ssl/ssl.h:75:
In file included from ../lib/axtls/ssl/tls1.h:45:
../lib/axtls/crypto/os_int.h:60:10: fatal error: 'endian.h' file not found
#include <endian.h>
         ^
1 error generated.
make: *** [build/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build/genhdr/qstr.i.last'
Works-MBP:unix work$ 
```

anna-liao, 2016-06-02:
To get past this error:

/usr/local/include:$ ln -s /usr/include/machine/endian.h endian.h

Anton-2, 2016-06-02:
endian.h moved to machine/endian.h since OS X 10.9.
And it's already included in sys/types.h, so there is no need to include it again in os_int.h : I can compile/link axtls and micropython without the `#include <endian.h>`.

It's something to fix in axtls, but I don't know how to reach them.
Maybe in @pfalcon's fork ?

stefanhoelzl, 2016-06-11:
Could you compile it now?
After I copied the endian.h file, I get the following error:
`LINK micropython
ld: library not found for -laxtls
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [micropython] Error 1`

thekrunchyfrog, 2016-06-14:
That was a pain.
I had the same problem. Removing the include from os_int.h worked.
Thanks @Anton-2 

The wiki on installing on mac should be udpated.

ilovezfs, 2016-07-11:
Upstream axTLS (http://axtls.sourceforge.net/) has now fixed this:
https://sourceforge.net/p/axtls/code/256/tree//trunk/crypto/os_int.h?diff=511c3f4934309d74f9522c0d:255

and has tagged a new version
https://sourceforge.net/p/axtls/code/259/tree/tags/release-1.5.4/

It is not yet fixed in the micropython fork, though. Cross-post here: https://github.com/pfalcon/axtls/issues/1#issuecomment-231686500

pfalcon, 2016-07-11:
If you can confirm that master builds ok on MacOSX, then yes, it can be closed.

ilovezfs, 2016-07-11:
@pfalcon master currently fails on macOS due to https://github.com/micropython/micropython/issues/2233

But if I do
```
inreplace "mpconfigport.mk", "MICROPY_PY_THREAD = 1", "MICROPY_PY_THREAD = 0"
```

master does indeed build without further patching.

ilovezfs, 2016-07-11:
@pfalcon now no patching is needed to build HEAD: https://github.com/micropython/micropython/issues/2233#issuecomment-231735329

Any chance of getting a new tag to simplify https://github.com/Homebrew/homebrew-core/pull/2607 ?

pfalcon, 2016-07-14:
@ilovezfs : Thanks for confirming! As for tags, @dpgeorge manages that, and while we don't have it as distinctive as  with Linux kernel, we usually have "freeze" period before a release (do, axTLS could not have been upgraded immediately before 1.8.2 release), and then after it, bunch of features land which need more development and testing, and that process has already started. That shouldn't be a big problem, as we have quite regular releases. So, please feel free to stay around and contribute more things for better MacOSX support. E.g. contributing .travis.yml would be useful thing ;-).

closed 2016-07-14 by pfalcon