micropython/micropython #2145: warnings in C code preventing SDK from compiling on Mac OS 10.11.5

anna-liao, 2016-06-02:
Going through README and building SDK, I encounter warnings after this process:

$ cd unix
$ make axtls
$ make

```
CC ../lib/timeutils/timeutils.c
../lib/timeutils/timeutils.c:175:34: warning: comparison of unsigned expression < 0 is always
      false [-Wtautological-compare]
    if ((seconds = seconds % 60) < 0) {
        ~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~
../lib/timeutils/timeutils.c:181:34: warning: comparison of unsigned expression < 0 is always
      false [-Wtautological-compare]
    if ((minutes = minutes % 60) < 0) {
        ~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~
../lib/timeutils/timeutils.c:187:30: warning: comparison of unsigned expression < 0 is always
      false [-Wtautological-compare]
    if ((hours = hours % 24) < 0) {
        ~~~~~~~~~~~~~~~~~~~~ ^ ~
../lib/timeutils/timeutils.c:194:30: warning: comparison of unsigned expression < 0 is always
      false [-Wtautological-compare]
    if ((month = month % 12) < 0) {
        ~~~~~~~~~~~~~~~~~~~~ ^ ~
4 warnings generated.
```

So to get pass this, commenting the line in unix/Makefile:

```
# compiler settings
# CWARN = -Wall -Werror
```

pfalcon, 2016-06-02:
@anna-liao : Thanks for the report.

@dpgeorge : I guess the correct way to fix these is to accept args to this function as mp_int_t, right? And this is critical for https://github.com/micropython/micropython/issues/2131

pfalcon, 2016-06-02:
Fixed in 993cc3611a991aaf2ac9066d0bec852de27eab9b.

closed 2016-06-02 by pfalcon

dpgeorge, 2016-06-02:
Yeah that looks correct.