micropython/micropython #2146: recursive_iternext test in test suite is failing on make test

anna-liao, 2016-06-02:
$ make test
1 tests failed: recursive_iternext

To run just the test by itself:

tests: $ ./run-test misc/recursive_iternext.py

I've tried running the first 2 sections in ./micropython interpreter and it doesn't throw a RuntimeError, as is shown in recusive_iternext.py.exp

try:
    [0] * 10000
    #N = 1000
    N = 10
except:
    N = 100

try:
    x = (1, 2)
    for i in range(N):
        x = enumerate(x)
    tuple(x)
except RuntimeError:
    print("RuntimeError")

Also, when changing the printout to "Runtime Error 1", the output in recursive_iternext.py.exp is still "Runtime Error"

pfalcon, 2016-06-02:
We have a CI setup, and as https://github.com/micropython/micropython shows, the build is green. So, this cannot happen in general. You probably mean a particular configuration you built, if so, please provide as detailed as possible information about it.

fatbird, 2016-06-02:
I see the same error.  We're using the unix build on OS X (El Capitan), on the current master branch, built per README instructions.  

One possibly relevant detail: the build failed due to warnings emitted from timeutils.c, and to complete the build we had to comment out -Werror in the Makefile.  The same warnings were emitted, and no other warnings.

Cabalist, 2016-06-02:
@pfalcon  This is the issue being referenced: https://github.com/micropython/micropython/issues/2145

Cabalist, 2016-06-02:
@pfalcon I just ran the tests with your latest commits.  On a clean Mac install I did the following:

```
$ cd micropython
$ git submodule update --init
$ ln -s /usr/include/machine/endian.h /usr/local/include/endian.h
$ brew install autoconf automake libffi libtool  pkg-config python3
$ export PKG_CONFIG_PATH=/usr/local/Cellar/libffi/3.0.13/lib/pkgconfig
$ cd micropython/unix
$ make deplibs
$ make 
$ make test
```

Result:
```
...
515 tests performed (17144 individual testcases)
514 tests passed
7 tests skipped: cmd_showbc cmd_verbose machine1 machine_mem urandom_extra buffered_writer extra_coverage
1 tests failed: recursive_iternext
make: *** [test] Error 1
```

Is there anything else about my environment that you'd like to see?

Anton-2, 2016-06-02:
When building with MICROPY_PY_USSL = 0 we don't need axtls, and no need to mess around with endian.h. But the recursive_iternext test still fails.

Looking at recursive_iternext, it seems to be checking that we throw a RuntimeError (instead of overflowing the stack, maybe). And the loop value used (N=1000) is not enough to make it reach this point on recent OS X/clang/??? combination.

Using N=10000, I get RuntimeError, and all tests pass.

dpgeorge, 2016-06-02:
> And the loop value used (N=1000) is not enough to make it reach this point on recent OS X/clang/??? combination.

That's probably the issue.  We have a fixed stack limit (something like 40k) on the unix build which might just allow more than 1000 recursions on Mac.  @Anton-2 can you please try N=2000?

Anton-2, 2016-06-02:
N=2000 is Ok, the four test cases give RuntimeError.
At 1500, only the first two test cases print RuntimeError.

dpgeorge, 2016-06-02:
Thanks.  I guess it should be changed to N=2000 then.

anna-liao, 2016-06-02:
I checked that the test passes for me with N=2000.

fatbird, 2016-06-02:
N=1625 is where it starts passing for me.

dpgeorge, 2016-06-03:
Fixed.

closed 2016-06-03 by dpgeorge