micropython/micropython #2154: WDT support

mangelajo, 2016-06-04:
Recently I got into issues with a faulty boards which eventually would freeze the processor,  on esp8266 platform I didn't have WDT,  so I had to move back to arduino esp8266 framework where the issue was alleviated that way (no more frozen electronics,  but may be a reboot a day which is ok) 

I believe we need WDT support for esp8266. 

Great work btw,.

dpgeorge, 2016-06-05:
AFAIK there is a system WDT thats running in the ESP8266, but the feeding is called automatically as long as your Python code is running (or you're at the REPL).

I guess what you are asking for is a user WDT, something along the lines of:
```
wdt = machine.WDT(timout_ms=1000)
wdt.feed()
```
Would this have helped your case?  What exactly was "freezing"?

Please see #264 for a related discussion.

mangelajo, 2016-06-06:
The mechanism you describe wasn't apparently working. The board was freezing after modifying a relay. 

The simplest reproduced was:

   while true:
          relay.high() 
          sleepms(500)
          relay.low () 
          sleepms(500)

Randomly the board froze forever and you had to power cycle it.

dpgeorge, 2016-06-06:
What is "relay", is it just a pin?  And is sleepms just time.sleep_ms?

nmistry, 2016-08-15:
Curious whats the official status of WDT support for ESP8266.   The docs (http://docs.micropython.org/en/v1.8.3/esp8266/library/machine.WDT.html)  show WDT support, but when i try to invoke via 1.8.3 REPL i get:

```
MicroPython v1.8.3-8-gaf9889f on 2016-08-12; ESP module with ESP8266
Type "help()" for more information.
>>> from machine import WDT
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name WDT
>>> import machine
>>> dir(machine)
['__name__', 'mem8', 'mem16', 'mem32', 'freq', 'reset', 'reset_cause', 'unique_id', 'idle', 'sleep', 'deepsleep', 'disable_irq', 'enable_irq', 'time_pulse_us', 'RTC', 'Timer', 'Pin', 'PWM', 'ADC', 'UART', 'I2C', 'SPI', 'DEEPSLEEP', 'PWR_ON_RESET', 'HARD_RESET', 'DEEPSLEEP_RESET']
```
maybe i am missing something?

dpgeorge, 2016-08-17:
The docs show a WDT but that's only implemented for the WiPy (cc3200 port).  In 64c5a9435c3b100fdc50daafebf8550ab5f40780 I added a note to this effect.

pfalcon, 2016-08-20:
Ok, so our reply to this was usually "WDT is automatically run by ESP8266  OS", but d'oh, I found that we have still a call to disable from early development times! So, I made sure it's not disabled in f2a21a2489325df88acf14a20a286932ed32d500.

pfalcon, 2016-08-20:
I'm also going to add some kind of machine.WDT class.

dpgeorge, 2016-08-25:
> but d'oh, I found that we have still a call to disable from early development times

Yes, but from my experimenting with it there's still a WDT somewhere that runs (that's why time.sleep(10) kept crashing the board, until the new execution model, ets_iter_loop, was implemented).  Are there multiple WDTs?

> I'm also going to add some kind of machine.WDT class.

+1  I guess it needs to be synthesised in software.

pfalcon, 2016-08-25:
> Are there multiple WDTs?

Possibly. I'll post my exact testcases later, but they were in the vein that without this change, machine.disable_irq() hangs them module, while with this change, will eventually reset it, with expected (WDT) reset reason.

> I guess it needs to be synthesised in software.

Nope, just provide access to the real WDT. Yep, it won't be much useful, but if people ask, why not... (My motivation is proceeding with ESP8266 RE).

dpgeorge, 2016-08-26:
> Nope, just provide access to the real WDT.

If there are indeed 2 "hardware" WDTs then, yes, we can provide access to one of them.  But remember that a watch dog is already fed in ets_loop_iter so it would be pretty useless to expose that one to the user (all it proves is that the VM is running, not that the user code is reaching their own wdt.feed() calls).

joseangeljimenez, 2016-08-26:
My two cents here:

- A watchdog making sure that the VM is running otherwise resetting the ESP will be of great help. This allows, for example, a simple script running in loop without worrying too much about uncaught exceptions. If there is an unexpected exception, the VM should stop and the WDT will make sure that the ESP is reset. In this scenario, a simple software switch to enable/disable this VM WDT will make this feature even more flexible (just in case somebody prefers that the ESP is not reset when the VM stops).

- If available in hardware, a second WDT offers additional measures for making sure an script is reaching the critical code sections. However, this will need additional work from the developers. IMHO, this is a less common scenario than the previous one.

Thanks for your great work!

pfalcon, 2016-08-28:
I've pushed initial code in 263aaa7.

dpgeorge, 2016-08-28:
> I've pushed initial code in 263aaa7.

I don't understand the motivation behind this implementation... I'd say it's more harmful having this than not, because users get a false sense of safety using this WDT.  Eg an infinite loop (`while True: pass`) will run without the WDT triggering because it's being fed by the VM loop, but the user would expect a reset be cause they never call wdt.feed().

pfalcon, 2016-08-28:
> I don't understand the motivation behind this implementation...

But enough was said above already. However, even just following criterion should be enough: HW API requires machine.WDT to be available.

> but the user would expect a reset be cause they never call wdt.feed().

The only expectation of classes in machine.* should be that they provide as direct as possible control of hardware. In particular, it was explained many times that addition of machine.WDT isn't going to help. People keep asking, so here goes machine.WDT.  It will acquire more capabilities over time. For example, now it can be fed. Next step is enabling/disabling it. Next step is setting timeout (rough details are known). In the very distant future it may be possible to un-make OS feed it, and leave that only to user app.

dpgeorge, 2016-08-29:
> The only expectation of classes in machine.* should be that they provide as direct as possible control of hardware.

They should also work as expected.

> People keep asking, so here goes machine.WDT.

Now people are going to ask "why doesn't it work as expected".  Please be prepared to deal with such bug reports.

pfalcon, 2016-09-03:
> Please be prepared to deal with such bug reports.

There should be (and will be, maybe not for this release) a section on WDT (as on everything else) in http://docs.micropython.org/en/latest/esp8266/esp8266/general.html , discussing port-specific peculiarities.

pfalcon, 2016-09-03:
In the meantime, .deinit() method implemented. So, now it's possible to disable WDT, but not re-enable it again.

pfalcon, 2016-09-04:
Promised demo of WDT in action:

Default state (WDT running):

~~~
>>> machine.disable_irq()
32
>>> [can't type anything here]
# after some time:
 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
...
MicroPython v1.8.3-107-g61e2dfd-dirty on 2016-09-04; ESP module with ESP8266
Type "help()" for more information.
>>> import machine
>>> machine.reset_cause()
1
>>> machine.reset_cause() == machine.WDT_RESET
True
~~~

With disabled WDT:

~~~
>>> machine.WDT().deinit()
>>> machine.disable_irq()
32
>>> [hangs forever]
~~~

phieber, 2016-10-28:
@pfalcon Any updates on this?
I have tried the latest nightly firmware as well as the stable. Same behaviour:

`from machine import WDT`
`wdt = WDT(timeout=2000)  # enable it with a timeout of 2s`
`Traceback (most recent call last):`
`  File "<stdin>", line 1, in <module>`
`TypeError: function does not take keyword arguments`

marci144, 2017-01-01:
@pfalcon Do we have any update on this subject?
I'm using normal ESP8266 chips not Pyboards and I still receive :
>>> wdt=WDT(id=-1,timeout=2000)                                                                                                                       
Traceback (most recent call last):                                                                                                                    
  File "<stdin>", line 1, in <module>                                                                                                                 
TypeError: function does not take keyword arguments 

or                                                                                                   
>>> wdt=WDT(timeout=2000)                                                                                                                             
Traceback (most recent call last):                                                                                                                    
  File "<stdin>", line 1, in <module>                                                                                                                 
TypeError: function does not take keyword arguments 

There is any work around using timers ? that the other option I can think of .... creating a timer that when triggered calls a machine.reset(), but it's not very elegant!
tks

ThomasRgbg, 2017-02-12:
Just so understand the status...  do I understand the code in micropython/esp8266/machine_wdt.c correct that only the id parameter is currently implemented, but no timeout or timeout_ms ?

kfricke, 2017-02-12:
Even better... you even seem to not be able feed it in your code. The ESP8266 OS does this for you. Only disabling interrupts can trigger it. So useless in 99% of real world use cases.

timpur, 2017-02-21:
Just wondering whats happening with the WDT for ESP8266 ? it doesnt seem to be working at all for me?
is this a feature to be fixed soon ? kinda want to use it in a project :P

pfalcon, 2017-02-23:
@timpur : Please re-read thread above, and you'll know that you don't need to worry about WDT, it all works automagically. However, people worried about WDT, and to give their worries the right direction, some initial control of WDT (well, just disabling it) and reverse-engineering of more WDT features (contributed to esp-open-rtos headers) were done, with the hope that interested people will continue that effort. So far nobody did, which objectively would mean that there's not enough interest in further WDT control.

I actually post this, inspired by an apparent change to that dead-lock: there's now a guy who both cares about MicroPython and does their reverse-engineering homework, meet @ARF1. See https://github.com/micropython/micropython/issues/2898#issuecomment-281957306 for more info.

So, anyone interested in WDT or any other advanced feature for esp8266, don't hesitate to follow @ARF1's example, reverse engineering is so much fun and you'll never be able to complain again that you don't have something interesting to do ;-).

@ARF1: You might be interested to look into WDT stuff too, it wasn't too faraway from being able to control WDT timeout, etc. (in principle, - esp8266 firmware has many "WDTs", so while you can control something, it may take extra effort to understand what exactly, and yet more to understand the whole picture - and granted, WDT isn't the most exciting thing to work on (but definitely simpler than many other, so may be good for flexing the skills)).

ARF1, 2017-02-23:
Off-topic addressed to all interested in ESP8266 reverse engineering:

@pfalcon I think a lot of the reverse engineering effort of the esp8266 (and esp32?) is hindered by the fact there is no central repository for reverse engineered code.

I often come across code snippets where somebody has reverse engineered a particular rom function they were interested in but they generally have two issues:

1. They are hard to find because they are in random, personal github repositories, deeply buried in forum entries, as comments in some code, etc.

2. They utterly lack auditability. You just have to trust that the decompiled code is correct. - A big ask considering how easy it is to make mistakes reverse engineering assemblies.

3. They gave up halfway, leaving their work unfinished and nobody knows that their work could be completed with little effort.

4. ESP8266 rom functions often reference locations in ram whose purpose is generally unknown. If reverse engineering of one rom function reveals the purpose of a particular ram location (e.g. pointer to idle callback function, ...) this can inform the reverse engineering effort for other functions: Now they are no longer referencing obscure ram locations but you already have an idea what some of these are.

What do you think about a esp8266 reverse engineering github repository/project in which:

1. Reverse engineered code is submitted as PRs in a standardised manner, e.g. 1. staring with a raw disassembly, 2. hand-annotated assembly, 3. c-code that follows assembly code logic, 4. refactored c-code. This would allow everybody to easily audit any reverse engineering effort for correctness.

2. A standard set of rom disassembly tools are provided which automatically annotate disassemblies with e.g. already known ram locations, etc.

3. Easy to follow instruction on how to get started on reverse engineering a new function. I think lowering the barrier to entry for novices to get started would go a long way to getting more people to help.

What do you think?

- Are there enough people interested to keep something like this alive? My gut feeling is that at least three persons would have to take ownership of such a project to ensure it stays alive (after all we all have jobs and other projects to work on).
 - Would people be willing to be told "how to work and what tools to use"?

pfalcon, 2017-02-24:
@ARF1: Thanks for the response. I posted a quick offtopic note like that because I didn't expect it to go into a bigger discussion. But if you made such a great write-up (really thumbs up), let's not pretend that it starts right here, right. There's already, and for a long time, a mailing list, where ESP8266 RE communication happens, my comment linked to it, and here it's again: https://groups.google.com/forum/#!forum/esp8266-re . So, I'd suggest to repost your comment as is to that mailing list.

> Would people be willing to be told "how to work and what tools to use"?

We'll see it right by this small usecase - whether the suggestion to move the discussion to the appropriate place will work ;-).

@trebisky: Are you subscribed? ;-)

mhv-shared, 2018-01-19:
> So far nobody did, which objectively would mean that there's not enough interest in further WDT control.

Or that the people interested in using MicroPython tend not to be the ones capable of reverse-engineering firmware. There is a clear use case for this: blocking sockets causing a lockup when connectivity is lost, e.g. http://stackissue.com/micropython/micropython/unstable-mqtt-on-esp8266-4-days-2568.html

dpgeorge, 2018-07-03:
In bccf9d3dcfd000b81b89b9ef862f9bc6104df99d I made it so the software WDT will cause a reset if it is not fed in time by the user script.  The timeout for this WDT is between 1.6 and 3.2 seconds and currently can't be changed.  To test use:
```python
w = machine.WDT()
w.feed()
# now wait for about 3 seconds for reset
```

pmth, 2018-09-03:
@dpgeorge I tried the watchdog today and it doesn't seem to be working on my 8266. Did I do anything incorrectly?

`
import time
from machine import WDT, Pin

def run():
    wdt = WDT()  # enable it with a timeout of 2s
    wdt.feed()

    print('Init')
    while True:
        print('Loop!')
        time.sleep(1)
`

peterhinch, 2018-09-03:
It's working here on the reference board. After pasting your code at the REPL:
```python
>>> run()
Init
Loop!
Loop!

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30944, room 16 
tail 0
chksum 0x4b
load 0x3ffe8000, len 1084, room 8 
tail 4
chksum 0xf5
load 0x3ffe8440, len 3252, room 4 
tail 0
chksum 0x55
csum 0x55
����n��r��n|�
               l
                lll`��r�l�l�

                              l`��r�l�l�

                                          l`��r�l
                                                  �

                                                   ll`rlrl����b
                                                                ��b�b
                                                                     br��OSError: [Errno 2] ENOENT

MicroPython v1.9.4-479-g828f771-dirty on 2018-09-02; ESP module with ESP8266
Type "help()" for more information.
>>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with misspiggy, channel 3
dhcp client start...
ip:192.168.0.19,mask:255.255.255.0,gw:192.168.0.1
pm open,type:2 0

>>> 
```

peterhinch, 2018-09-03:
I can't replicate this: the above script works fine, with the device recovering normally from the reset when it eventually occurs.

This is the MicroPython bug tracker: general programming support requests should be raised in [the forum](https://forum.micropython.org/) where they will reach a wider audience.

dpgeorge, 2021-04-14:
The esp8266 has a working WDT, so I'll close this issue.

closed 2021-04-14 by dpgeorge