micropython/micropython #2161: MICROPY_BUILD_DATE makes builds nondeterministic

MarkusTeufelberger, 2016-06-05:
This means a binary I create today will be different from one I create tomorrow using the exact same code + compiler + toolset.

stinos, 2016-06-06:
Only when your compiler/toolset doesn't already generate different binaries (like, including timestamp) every time the exact same code gets compiled. Pretty sure for example gcc and cl do that by default.

dhylands, 2016-06-06:
IIRC COFF files (the default used under Windows) include a timestamp, but ELF files (the default under linux) do not.

For the times that it matters (typically some type of automated test), I normally write a utility that zaps the timestamp with a known good value.

That still doesn't help when the source code includes things like __DATE__, but that's a different issue.

dpgeorge, 2016-06-06:
Calling it "nondeterministic" is a bit of a stretch.  Why is it important for the binary to be equivalent?

MarkusTeufelberger, 2016-06-06:
https://wiki.debian.org/ReproducibleBuilds - if you search for "reproducible builds" e.g. on youtube you'll find a few talks by members of this initiative that explain better than I could why this is a desired/important property to have in build artefacts. Why is the build date displayed on startup?

dpgeorge, 2016-06-06:
> Why is the build date displayed on startup?

Following CPython.  It could arguably be changed to the date of the commit, not the build.

jhdulaney, 2016-06-09:
Reproducible builds are a security feature to allow for checking against malicious code insertion.  Time of commit instead of time of build would be the better way to go.

pfalcon, 2016-06-09:
@MarkusTeufelberger, @jhdulaney : Sorry to put it like that, but you guys bark on a wrong tree. You should go and find that debian wiki page which rants how it's cool to have that .gnu.buildinfo or how it's called section with a unique hash for each executable, to help NSA trace the source of each in the world. Now there's page which says the opposite. I assume, I didn't read it. I'll wait for wiki page which will tell that address space randomization is bad for security and read all them together, ok? Because the above must be true, randomization is more code, a code is buggy => bigger attack surface.

Moral: when the world will stop putting useless bloated random stuff in executables and around them, we will follow. In the meantime, we have better things to do than apply and unapply random things back and forth, following the latest niche ~~whims~~ trends.

MarkusTeufelberger, 2016-06-09:
ASLR is randomization at run time enforced by the user's OS, this is randomization at build time enforced by the compiler of the person who built the code.

I agree that you have lots of better things to do, this was just a quirk that I stumbled upon while reading your code and I remembered that having compile dates is rarely a desired feature or necessary/helpful. As per your contribution guidelines, I just opened a ticket here to discuss this finding, since I didn't want to sign up for yet another forum account. Sorry, if it was the wrong place.

If you are concerned about bloat, how about removing this field alltogether?

ncoghlan, 2016-06-09:
As far as the CPython comparison goes, we should likely also be tracking the source commit date rather than the build date - that's a pretty common approach to filtering out extraneous build details (iike precisely when you can the build command) from significant ones (like exactly which source code you used).

I think most of the reproducible build focused work is currently happening downstream in Linux distros rather than upstream in the main CPython tree, though.

jhdulaney, 2016-06-09:
Reproducible builds is one of those things that ought to be a joint partnership between downstream and upstream.  While, yes, it is mostly a downstream effort (with toolchain and the like), sometimes upstream involvement is also necessary.

pfalcon, 2016-06-09:
> I remembered that having compile dates is rarely a desired feature or necessary/helpful.

You must be kidding us, right? *Everyone* puts compile time in binaries. And my point is: feel free to go convert dozen of other projects first, and share experiences how well it went. Feel free to start with these dudes:

~~~
$ uname -a
Linux x230 3.13.0-66-generic #108-Ubuntu SMP Wed Oct 7 15:20:27 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
~~~

> I just opened a ticket here to discuss this finding, since I didn't want to sign up for yet another forum account. Sorry, if it was the wrong place.

You're welcome to, but if you expect some resolution, well... you'll get the same response as from any other project to which random changes from unicorn reality is proposed.

To give this productive direction: it's up to @dpgeorge, but I don't see any defaults changing. However, if you're interested to make it possible to diff disassembly output a little bit better, feel free to propose simple and clean changes which allow to pass something on make command line to achieve that (assuming it's not achievable already).

MarkusTeufelberger, 2016-06-09:
You mean https://tests.reproducible-builds.org/debian/rb-pkg/unstable/amd64/linux-latest.html ?
Since this is apparently "unicorn reality" and something that _everyone_ does anyways, I'll just leave it at that.

Personally I'm more involved in projects that do _very_ much care about verifiable builds (such as bitcoin or tor), I just stumbled upon this one here when looking for an interesting target to try out libfuzzer and because I kinda like Python. Feel free to review my actual contribution to this project in a different pull request and/or close this issue once you've reached a decision.

meski, 2016-06-09:
If you want timestamps to be really useful, you need to use something like
https://en.wikipedia.org/wiki/Trusted_timestamping   - where the contents
of the file are hashed.  Being able to read these from within the app might
take up most of the microprocessors memory, so would be overkill on some
systems.

-- 
Meski

"Going to Starbucks for coffee is like going to prison for sex. Sure,
you'll get it, but it's going to be rough" - Adam Hills

pfalcon, 2016-06-10:
> You mean https://tests.reproducible-builds.org/debian/rb-pkg/unstable/amd64/linux-latest.html ?
Since this is apparently "unicorn reality" and something that everyone does anyways, I'll just leave it at that.

This gets completely offtopic, but as we have such a nice chat here: yeah, you understood it right - that's unicorn reality. And debian goes right into it since it adopted systemd. A week ago I tried to find a wiki page on debian way configuring wifi connection and found literally nothing useful. But at least there's a page on "reproducible builds".

> (such as bitcoin or tor)

Cool, love both! But just because I work on MicroPython, I don't go to them and tell that they should target the minimal code size and run on microcontrollers.

> Feel free to review my actual contribution

All contributions are appreciated and welcome, and will be reviewed with due diligence. However, you may find that contributions along the lines of:

"Hey guys, I found 10kg more weight for you to carry! Now I grip it firmly to carry it along with you."

are *slightly more* appreciated than ones along the lines of:

"Hey guys, I found 10kg more weight for you to carry! Squat, so it's more comfortable for me to put them on your back. Feel the weight? It's real 10kg, not some piece of cake! Now let me stroll away, maybe our paths will cross again some day!"

So, if you found some real bug with you fuzzer and fixed it without bloating uPy code, don't hesitate to share it. Otherwise, thank you very much too, we'll get to look into it some day.

MarkusTeufelberger, 2016-06-10:
I don't even use Debian or related distros... they are just one of the more vocal proponents of this issue and I thought it might be useful to you if you want to read up on the topic in case some of you might not be familiar with it.

It seems to me that you are upset for some reason with the Debian project and subsequently me as someone who posted a link to it. So far you have only complained about bloat and gave some very generalized statements without any actual constructive actions or arguments (e.g. "build dates need to be displayed at startup because in our case we use them to..."). Just as general feedback: To me this does not look encouraging as a new potential contributor, as your language seems very close to attacking the person, not the issue.

My suggestion is to either remove the build date alltogether (=less code + smaller binary size) or change it to the date of the referred commit (=more useful date + more human readable and accessible than just a commit hash).

> So, if you found some real bug with you fuzzer and fixed it without bloating uPy code, don't hesitate to share it.

Oh, I found lots of them, the main question for me is if the issue is with the fuzzer or uPy. This is why I started by sharing the fuzzer and not the bugs, as they might be false positives.

> Otherwise, thank you very much too, we'll get to look into it some day.

Great. :-)

pfalcon, 2016-06-10:
> It seems to me that you are upset for some reason with the Debian project and subsequently me as someone who posted a link to it.

If that's the only thing you grasped from the conversation, it's sad. Because what I tried to convey is that there's no need for everyone in the world to fall to today's whim of a particular project, even if that project is Debian (which I use, thru Ubuntu, in preference to all other Linux distros).

> as your language seems very close to attacking the person, not the issue.

Sorry, I suspect you don't completely follow thru my metaphors, but he's another one: I'm riding bicycle, you stop me and tell that I should ride a car. I tell that I'm fine riding bike, you tell that I still should ride a car, everyone rides a car (which is false), big guy Debian rides a car. I use stronger arguments to deter you from stopping random people who don't care about cars, you call me using language. Who's right, who's wrong?

> "build dates need to be displayed at startup because in our case we use them to..."

I thought you're versed enough in open-source process, because the answer to such questions is always the same: "Because an earlier project member contributed such code, and we won't change it just because a *potential* contributor came along and tells us do that. Because if we do, a next guy after you will tell us to change *your* code, and it will be gone or put upside-down without a good reason."

> without any actual constructive actions or arguments

Let me repeat:

> To give this productive direction: ... However, if you're interested to make it possible to diff disassembly output a little bit better, feel free to propose simple and clean changes which allow to pass something on make command line to achieve that (assuming it's not achievable already).

MarkusTeufelberger, 2016-06-10:
Do you prefer the removal or the commit date option or do you not care (then I'd go with the removal, as it removes bloat)?

pfalcon, 2016-06-10:
All questions should have been answered already:

>  it's up to @dpgeorge, but I don't see any defaults changing

defaults == how it's being done now by default

Mind that as everything else I write, it's just my personal opinion.

MarkusTeufelberger, 2016-12-11:
So now that a patch is available for 5 months, did you find the time to review it?

dpgeorge, 2016-12-14:
> So now that a patch is available for 5 months, did you find the time to review it?

@MarkusTeufelberger sorry but this is just a really low priority issue, hence no feedback on it yet.

My feeling on the issue is the following: having build date gives more information than having commit date.  I can find out the date of a version by looking up the code, and having the build date there as well tells me something more.  For example, if I pick up a random board that has uPy on it then I can see when I last compiled/deployed the firmware by the date in the banner.  When developing code this extra info is more important than deterministic builds.  From the end-user point of view, the build date is generally the same as the date of the commit of the release because release firmware is automatically built twice a day.

closed 2016-12-14 by dpgeorge