micropython/micropython #2164: Compiling for esp8266; I'm not able to use main/boot.py

DavidVentura, 2016-06-06:
This is pretty basic; I've flashed my esp8266 a few times and I get the REPL console.
But adding a main.py / boot.py in the scripts directory doesn't do anything. My board doesn't have USB so I can't really add them to the board itself directly.
What should I do to get the board to parse my scripts? I've been manually pasting them in a single line with newline characters in the console.

dpgeorge, 2016-06-06:
Try using the webrepl_cli.py script from this repo: https://github.com/micropython/webrepl .  It allows you to upload/download scripts over wifi (using webrepl).

deathmock5, 2016-06-07:
I have main and boot in there, but I also cant get them to execute at boot.

robert-hh, 2016-06-08:
After adding files to the scripts directory, you have to rebuild the image to make them accessible, using:
```
make clean
make
```
SInce the make file does not detect changes to the scripts directory, make clean is required,

452, 2016-06-09:
Since the make file does not detect changes to the scripts directory, make clean is required = also have a problem with this, with intuition, I also just try make clean.

```sh
MicroPython v1.8.1-19-gbba77a2 on 2016-06-08; ESP module with ESP8266
Type "help()" for more information.

>>> os.getcwd()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV

>>> os.listdir()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV

>>> os.mkdir('/kot')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```

I try to run https://github.com/RinusW/WiPy/tree/master/AiCWebserver
but have a problem with
```python
with open('AiCWebpage.htm', 'r') as html:
```
/git/micropython/esp8266/scripts/AiCWebserver.py
/git/micropython/esp8266/scripts/AiCWebpage.htm

please, can you provide some documentation about internal file storage in ESP8266?
or how to make a simple web server but with html page from file

DavidVentura, 2016-06-09:
I've had this issue before. Erase the flash and try flashing at 115200. You seem to not be getting a valid filesystem

pfalcon, 2016-06-12:
Hopefully the responses above answer the original question. There're also boot process docs at http://docs.micropython.org/en/latest/esp8266/esp8266/general.html#boot-process . (Docs also cover many other important and useful topics.)

closed 2016-06-12 by pfalcon

wavesailor, 2017-02-14:
I had the same issue as originally mentioned. I'm using an ESP8266 board (https://www.adafruit.com/product/2471). Seemed like the file system was corrupt or not loading. 

What I did to fix it was to enable the AP
`ap = network.WLAN(network.AP_IF) # create access-point interface`
`ap.active(True)         # activate the interface`
`ap.config(essid='ESP-AP') # set the ESSID of the access point`

After rebooting I could then access the file system with
`>>> import os`
`>>> os.listdir()`

Hope  this helps someone else as it took me ages to solve it.