micropython/micropython #2171: esp8266 automatically connects to wifi. Why? How?

DavidVentura, 2016-06-07:
I've been playing with this on the REPL console and connected to my wifi. On reboot it automagically connected even though there are no files on the filesystem. This is cool but I'd like it to be consistent. How is it working right now?

dmarkey, 2016-06-07:
I'm pretty sure the esp8266 SDK saves the wifi config on EEPROM internally so that changes are persistent across reboots.

vmesel, 2016-06-08:
The ESP8266 saves your wifi info, even after soft-resetting it, it will continuously be available. As I reported in the #2141 bug, it keeps trying to reach out my wifi.

pfalcon, 2016-06-12:
@DavidVentura : As hinted in previous replies, that's just how ESP8266 works. Everyone who has experience with ESP8266 knows that and expects that. If you're just starting with ESP8266, please be sure to read the vendor documentation to get acquainted with how it works in general. We also have a great forum for user support: http://forum.micropython.org .

closed 2016-06-12 by pfalcon