micropython/micropython #2174: Support openssl in addition to axtls

jhdulaney, 2016-06-09:
For use cases where micropython is built for inclusion in a distro that is probably going to have openssl installed by default, it may not be a bad idea to support openssl.  I'm willing to do the work to do so, but I don't want ot proceed unless this is considered to be acceptable.

pfalcon, 2016-06-09:
If you drop a nicely done code on us, it will be hard to reject it, but that's completely against micropython idea. Next TLS library to integrate with uPy would be mbedTLS, not openssl which is more binary size than uPy itself.

jhdulaney, 2016-06-09:
The other issue is the last time axtls was updated was a year ago.  For something involving ssl, I'd think a more active upstream would be desirable due to the inherit security nature of ssl.

ncoghlan, 2016-06-09:
While I agree OpenSSL isn't interesting for the MicroPython-without-an-OS use case, MicroPython-on-Linux has emerged as an interesting alternate deployment option, especially in the context of minimalist distros like busybox or Alpine, as well as in stripped down versions of traditional distros that omit a full CPython interpreter.

For those latter cases, integrating with OpenSSL offers a potential means of integrating with the system certificate stores.

Any such additions would naturally be optional and/or specific to the *nix builds.

pfalcon, 2016-06-09:
> The other issue is the last time axtls was updated was a year ago.

axtls isn't feature-reach (missing baseline features nowadays like SNI) or well-maintained library, however it's the smallest TLS lib with liberal license (likely, the smallest of open-source). And it was selected per criteria in https://github.com/micropython/micropython-lib/issues/69 (to make possible to work with HTTPS as if it was HTTP).

pfalcon, 2016-06-09:
Ok, so an implementation path which I'd suggest: don't touch existing extmod/modussl.c , make extmod/modussl_openssl.c (well, maybe it will be extmod/modssl_openssl.c). I guess it's also should be selectable by the existing MICROPY_PY_USSL, and another config param introduced to select which lib to link with.

jhdulaney, 2016-06-15:
What do y'all think about using pyopenssl as an option?

ncoghlan, 2016-06-15:
@jhdulaney As in, getting that running on the *nix variant of MicroPython and using it for the integrated *nix MicroPython SSL stack? I think that would be an *excellent* option if you can make it work - there's no reason to duplicate PyCA's work if you don't have to. That relies on the cryptography module and cffi under the hood, which would presumably make supporting the latter the main integration challenge.

tedder, 2017-06-07:
FWIW, axtls has a lot of recent updates.

pfalcon, 2017-06-07:
@tedder : Patches welcome ;-)

pfalcon, 2017-06-14:
MicroPython now supports two separate "ussl" module implementations: using axTLS and using mbedTLS. This should pave the road to support implementations against any other libraries, though question of testing and maintenance of additional implementations remains.

Closing the ticket otherwise.

closed 2017-06-14 by pfalcon