micropython/micropython #2175: open socket and webrepl at the same time

DavidVentura, 2016-06-10:
I've implemented a socket that allows the device to be pseudo remote-controlled. I can get/set gpio values and reset the device. To get 2 more GPIO I've stopped using serial to get a terminal and started using webrepl.

This works if I run `webrepl.start()` and then manually `server()` (the server call blocks, but I can interrupt it and continue using the shell)

Now I want to run the `server()` on boot and launch webrepl on demand. Is there a way to do this?

pfalcon, 2016-06-12:
Thanks for submission to the MicroPython bug tracker. This bug tracker is intended to track confirmed bugs in the MicroPython codebase. However, your submission doesn't look like bugreport. For questions, support requests, suggestions, etc. you may find that using the forum gives better result:

forum.micropython.org

You can get even better results by first searching for existing topic where a similar issue may be discussed (this is also a good etiquette on any Internet forum, as otherwise you may end up asking a question which was answered just a day or an hour ago).

Alternatively, MicroPython developers may review your submission in more detail at later time, subject to the number and priorities of confirmed bugreports. This may involve up to a month, or more, waiting times.

pfalcon, 2016-08-09:
@DavidVentura : As suggested above, please try to post your question to the forum at http://forum.micropython.org , you likely will find larger audience. You're welcome to post bugreports here though!

closed 2016-08-09 by pfalcon