micropython/micropython #2177: Expose the font from extmod/modframebuf.c to python programs

deshipu, 2016-06-10:
I wonder if it would be possible to have the font that is compiled into the framebuf module exposed to the python code somehow, so that it could be used by drivers that don't use a 1-but framebuffer -- like the drivers for color displays. For instance, a function that takes a character and returns a memory view with the relevant font data? Not sure how feasible or hard it would be to do, and if there would be an additional cost involved with that, but it seems to me a waste when each driver includes their own font and struggles to fit it in RAM.

deshipu, 2016-06-10:
Right now I do it with a dirty hack -- I create a 8x8 framebuffer, print the character to it, and then copy the character data -- but I think this could be done nicer?

https://bitbucket.org/thesheep/micropython-ili9341/src/4fe3cd28006e171eec16aff9e7cbae8ed394f88d/ili9341.py?fileviewer=file-view-default#ili9341.py-142:157

mcauser, 2016-06-30:
The framebuffer font is located at /stmhal/font_petme128_8x8.h
Each byte is represents a column, left to right, with the msb at the bottom.

Sharing the font data with other display libraries sounds beneficial.
A function on the framebuf module for returning a byte array representing the supplied character(s).

pfalcon, 2016-07-02:
So, exposing it pretty easy from technical point, but no so easy from API design point of view. The only option I see for now is to expose it as a private, undocumented function, likely to be gone later.

deshipu, 2016-07-02:
What would it be replaced with?

pfalcon, 2016-07-02:
By a text() function which just works.

deshipu, 2016-07-02:
How would that work? I'm eager to to try and write that, but my limited imagination doesn't let me think about something that would work in all possible cases with all possible third party drivers and libraries.

closed 2021-06-02 by deshipu