micropython/micropython #2178: how to make Combination key in USB_HID?

xiaowuyi, 2016-06-11:
## I want to make a Combination key such as shift+a,but i dont know how to do.
## I try like this:

#boot.py
import machine
import pyb
#pyb.main('main.py') # main script to run after this one
#pyb.usb_mode('CDC+MSC') # act as a serial and a storage device
pyb.usb_mode('CDC+HID',hid=pyb.hid_keyboard)

#main.py
hid=pyb.USB_HID()
buf=bytearray(8)
buf[2]=0xE1 #left shift
hid.send(buf)#press left shift
pyb.delay(10)
buf[2]=0x04#a
hid.send(buf)#press a
pyb.delay(10)
buf[2]=0
hid.send(buf)

## but it cant work.how can i make a Combination key?

peterhinch, 2016-06-12:
I've responded to your post in the forum. GitHub is intended for raising issues such as bug reports in the MicroPython language itself, rather than being a place for requests for programming guidance.

xiaowuyi, 2016-06-12:
ok,i am sorry,I didn't expect to bring you so much trouble.sorry!

closed 2016-06-12 by pfalcon