micropython/micropython #2179: Can not connect to WiFi, exception

iSevenDays, 2016-06-11:
load 0x40100000, len 31696, room 16 
 rl l��| �l�|�l�b|����r�blb��nn�lnn���bp�lrlrlp�n��lbn�|�l��|�b��nn� l��l`�nnl`nr���n�b�`p�n�r�����bn�|l�b��nn� l`�nnl`nr���n��l`p�n�r������p�bn�|�ll�b��nn� l`�nnl`nr���nbll`r��nbll`l��|��rrnb��l��b�lb������l�b�lrlll��|��rrnb��l���b�b�����bb��lrl�l��|��rrnb��l�r��pp�b������b�lblrl��rl���bl�br|�b�������l ��n�nn�ll����l��l����ll`�n� ���b��b���bl�b�lrlr�n�� ���n����brb��b쎜��l�bll�lrl��ln�prl�l��|��rrnb���could not open file 'boot.py' for reading
could not open file 'main.py' for reading

#4 ets_task(40100350, 3, 3fff6350, 4)
MicroPython v1.8.1-31-g9de5eb2-dirty on 2016-06-11; ESP module with ESP8266
Type "help()" for more information.
>>> 8*8 \nhelp()8*8d\nf
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 8*8
64
>>> print('hello esp8266!')
hello esp8266!
>>> import network
>>> sta_if = network.WLAN(network.STA_IF)
>>>  sta_if.active(True)
Traceback (most recent call last):
  File "<stdin>", line 1
IndentationError: unexpected indent
>>> sta_if.active(True)
mode : sta(5c:cf:7f:13:9b:95) + softAP(5e:cf:7f:13:9b:95)
#5 ets_task(4020e1cc, 28, 3fff7f28, 10)
add if0
>>> sta_if.connect('Apple', 'ervster1102')
f r0, >>> scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with Apple, channel 7
dhcp client start...
ip:192.168.76.6,mask:255.255.255.0,gw:192.168.76.1
Fatal exception 0(IllegalInstructionCause):
epc1=0x40209be0, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

pfalcon, 2016-06-12:
Looks like duplicate of https://github.com/micropython/micropython/issues/1946 . Please be sure to follow the latest post in http://forum.micropython.org/viewtopic.php?f=16&t=1689 and try again with the latest SDK. If it won't work still, feel free to report the issue to Espressif.

iSevenDays, 2016-06-12:
@pfalcon 

Thank you, I will try and then post my results

iSevenDays, 2016-06-13:
@pfalcon one day of testing with no problems,

You comment should be definitely added to the FAQ

Thanks!

closed 2016-06-14 by iSevenDays