micropython/micropython #2181: libaxtls.a: error adding symbols: File format not recognized

patvdleer, 2016-06-12:
In an attempt to compile the ESP version I did the following;

Followed the instructions from https://github.com/pfalcon/esp-open-sdk This was successful after which I tried to compile microPython, the Unix version does compile(make) and install but the esp8266 version doesn't.

history
~~~
git clone https://github.com/micropython/micropython.git
cd micropython/
cd unix
make axtls
make deplibs
make
sudo make install
cd ../esp8266/
make
~~~

Error:
~~~
CC ../lib/netutils/netutils.c
CC ../lib/timeutils/timeutils.c
CC ../lib/utils/pyexec.c
CC ../lib/utils/pyhelp.c
CC ../lib/utils/printf.c
CC ../lib/fatfs/ff.c
CC ../lib/fatfs/option/ccsbcs.c
CC ../drivers/dht/dht.c
LINK build/firmware.elf
../lib/axtls/_stage/libaxtls.a: error adding symbols: File format not recognized
make: *** [build/firmware.elf] Error 1
~~~

dhylands, 2016-06-12:
I haven't built for esp8266 yet, but if I had to hazard a guess, I would say that the make axtls step you did while in the unix directory will have built an x86 version of libaxtls.a, which is incompatible with the esp8266 architecture.

patvdleer, 2016-06-12:
Indeed, awesome gut feeling! Thanks!!!

~~~bash
cd lib/axtls/
make clean
cd ../../esp8266/
make axtls
make
~~~

closed 2016-06-12 by patvdleer