micropython/micropython #2182: Hardware API: Clarification to class naming and mandatory constructor arguments

pfalcon, 2016-06-12:
If you look at https://github.com/micropython/micropython/wiki/Hardware-API#the-i2c-class---i2c-master-end-point , https://github.com/micropython/micropython/wiki/Hardware-API#the-spi-class---spi-master-end-point , https://github.com/micropython/micropython/wiki/Hardware-API#the-timer-class , and finally https://github.com/micropython/micropython/wiki/Hardware-API#general-conventions-of-the-api , they all clearly say that the first mandatory argument for any HW peripheral class should be "id", with values 0 and higher reserved for actual hardware peripheral, values -1 and below for "virtual" peripherals.

Let's look at the situation with esp8266:

* Timer class follows the spec above, with my effort.
* Bitbanging I2C and SPI implementations - do not.

Given #2170, #2172, we need to resolve this now. My initial motion motion was to just follow the spec above, it was made to be followed after all.

However, thinking a bit more, given my argumentation for *possibility* of multiple independent duck-typed Pin classes, extending that logic to other HW API classes, there shouldn't be a single SPI or I2C class per port. There can be e.g. BitbangSPI and SPI (the latter HW one). But it's still subtle matter with my "multiple independent duck-typed Pin classes" proposal, I don't really call to *have* multiple classes, just for *possibility* of. The idea is that user should be able to provide standalone, yet efficient HW classes, not for example having Pin0_15 and Pin16 classes on ESP8266. Under my idea, "officiall support port hardware" will still be accessible via Pin(), which thus may be (or may be not) a factory function which will instantiate a particular underlying class  - and exactly based on "id" param.

So, based on that, I propose specifically for esp8266:

1. There's single SPI() function, with a mandatory first "id" param.
2. For id == -1, it will instantiate bitbanging SPI class (as used currently).
3. For id == 0, 1, it will instantiate HW SPI class developed by @deshipu .
4. Internal bitbang SPI/HW SPI classes doesn't need to be exposed directly.

deshipu, 2016-06-12:
I'm not entirely sure what advantage does enumerating all possible interfaces of a given type give, other than one specific use case of a program that is supposed to list all possible interfaces. Personally I find it much more convenient to use class names that relate to the names of the actual underlying hardware and appear in its documentation, especially when they may support different features, require different arguments and even work in different ways (for instance, it only makes sense to have one HSPI class, but you can have any number of software SPI classes).

pfalcon, 2016-06-12:
> I'm not entirely sure what advantage does enumerating 

I'm not sure what enumeration you mean. This ticket talks about identification/instantiation.

> the names of the actual underlying hardware and appear in its documentation

Hardware vendors are known to use idiotic names. Few examples: some Chinese company uses "HSPI" for (supposedly) "user SPI", "SLC" for "DMA", and "buad" for "baud". Many western companies use "TWI" for "I2C", etc. In MicroPython, SPI is SPI, I2C is I2C regardless of what marketing and pseudo-legal gimmick a vendor went for.

deshipu, 2016-06-12:
>I'm not entirely sure what advantage does enumerating

>> I'm not sure what enumeration you mean. This ticket talks about identification/instantiation.

Enumeration as in 1, 2, 3, 4, 5... you know, assigning integers to objects. I'm not convinced that numbers are the best way to refer to everything.

> Hardware vendors are known to use idiotic names. Few examples: some Chinese company uses "HSPI" for (supposedly) "user SPI", "SLC" for "DMA", and "buad" for "baud". Many western companies use "TWI" [...]

If we can't trust the makes of the hardware in question, who can we trust?

deshipu, 2016-06-12:
What about the slave interfaces? Are they going to be included in that? You might need to switch to complex numbers at some point.

pfalcon, 2016-06-12:
> Enumeration as in 1, 2, 3, 4, 5... you know, assigning integers to objects. 

I see, well, that's how hardware is made - is there's more that one peripheral block of the same type, it's copy of the same block, in pathological cases patches somehow. Most sane vendors assign numbers to them right away, the most sane even starting from 0. Others start from arbitrary numbers, like 1. A bit more creative - A, B, etc. Completely out of mind may call first instance SPI and 2nd - IPS. That all doesn't change anything - they are similar blocks which indeed can be enumerated.

> I'm not convinced that numbers are the best way to refer to everything.

Welcome to computer science - here, numbers are used to represent everything.

> If we can't trust the makes of the hardware in question, who can we trust?

In general sense, the general community in the area. As mentioned above, all vendors do it the same way, regardless how their product management and marketing departments twist description of it afterwards. In the case of specific vendor, we can trust independent parties which have no interest in upholding vendor b/s, and interested in representing hardware description as it is on hardware level. For esp8266, one of the best projects in this area (despite all its drawbacks) is esp-open-rtos. But you can see the truth even in MetalPhreak's driver version - there're to hardware SPI blocks in esp8266, and they are the same.

> What about the slave interfaces? Are they going to be included in that? 

This topic were discussed in 300+ message https://github.com/micropython/micropython/issues/1430 thread (and actually the questions above were touched there too), you're welcome to get acquainted with it.

torwag, 2016-06-13:
Being late to the party, I am not sure whether I can completely follow the arguments.
However, I like @pfalcon s number scheme for the sole purpose of being hardware independent,  SPI(0) will be an hardware SPI on the esp8266 as well as on any future board. This makes (micro)python code rather hardware-agnostic in the easiest case you could simply copy it from one to the other board.
On the other hand I see @deshipu argument that it takes extra brain power, and with the lack of it, possible bad user experience,  to translate something (as bad as) "USR_SPI_A" given in a datasheet to SPI(0) in micropython. People will frequently mix up and being confused. Docs need tables with mappings, etc.

Would it be possible to add an hardware-mapping dictionary and with the help of that enable to give an id, either by a number or the vendor name?
Hence SPI("USR_SPI_A") would simply be internally translated to SPI(0) 
Having a centralized dictionary, mapping ids to vendor_names (not only for SPI) makes maintenance relatively easy. Users can look up the dictionary to find the needed relation.

deshipu, 2016-06-13:
Or there could be ``HSPI`` and ``SoftSPI`` classes, which you can use directly if you want, and a factory function ``SPI`` that instantiates the one or the other, depending on the number. I think that's what @pfalcon is hinting at, but I'm not entirely sure.

Having one function that refers to different implementations, or even does completely different things, based on a parameter passed to it (like here SPI(0) uses one class, SPI(1) another one, or like in the sensors discussion, where you would get readings from different sensors depending on the sensor number you pass) is a C-ism, a pattern necessary in a statically typed and low-level language like C, but completely unnecessary in a dynamic language like Python, where you can easily pass around references to functions and methods, keep them in dictionaries, etc.

dpgeorge, 2016-06-14:
Note also the following from the Hardware API wiki page:

> Peripherals should provide default values for all the initialization arguments, this way when calling .init() with no params, or the constructor with only the peripheral id (or no id at all, then the default one is used), it will be initialized with the default configuration.

So it's allowed to not specify an id parameter at all, and in that case it'll chose the default (eg RTC() would be RTC(0)).  And if you specify pins without the id then it picks the default peripheral associated with the given pins.  In the case of the I2C and SPI classes on the esp, it currently always picks the software (bit-bang) implementation for all pin combinations that are passed in.

> So, based on that, I propose specifically for esp8266:
> 
>     There's single SPI() function, with a mandatory first "id" param.
>     For id == -1, it will instantiate bitbanging SPI class (as used currently).
>     For id == 0, 1, it will instantiate HW SPI class developed by @deshipu .
>     Internal bitbang SPI/HW SPI classes doesn't need to be exposed directly.

This sounds good.  But we can make the id default to -1 to conform to the point above about optional id (and also retain backwards compatibility).

deshipu, 2016-08-05:
Should I try to make such a factory function to push this a little bit forward?

pfalcon, 2016-08-06:
@deshipu : Sure, please do.

deshipu, 2016-10-12:
I think this can be closed.

closed 2016-10-12 by pfalcon