micropython/micropython #2184: unix: type of generator function is incorrect

peterhinch, 2016-06-14:
MicroPython seems unable to distinguish a generator function from a generator:
```python
def foo(x):
     while x > 0:
         yield x
         x -= 1
```
This correctly produces the following under cPython 3.4.3
```
>>> bar = foo(5)
>>> type(foo)
<class 'function'>
>>> type(bar)
<class 'generator'>
>>> 
``` 
Under MicroPython v1.7-305-g56fd33a-dirty on 2016-05-03; linux version
```
>>> bar = foo(5)
>>> type(bar)
<class 'generator'>
>>> type(foo)
<class 'generator'>
>>> type(foo) is type(bar)
False
>>> 
```

dpgeorge, 2016-06-14:
In uPy, generator functions are different from normal (non-generator) functions.  And different again from the activated generator object.  It's just that we use the same string/name for the generator function as the generator object. (namely "generator").

Is it important to have `type(gen_fun) is type(fun) == True` in uPy?  If so it requires structural changes to how generator functions are handled (and may actually lead to code size reduction..?).

peterhinch, 2016-06-14:
I observed this difference in behaviour when I needed to detect whether a generator or a generator function was passed as an argument. In MicroPython I found I had to explicitly code
```python
def _g():
    yield 1
GeneratorType = type(_g())
```
I could then do
```
>>> bar = foo(5)
>>> type(foo)
<class 'generator'>
>>> type(bar)
<class 'generator'>
>>> type(bar) is GeneratorType
True
>>> type(foo) is GeneratorType
False
>>> 
```
So the workround is simple, but perhaps unnecessary? Am I missing something here?

pfalcon, 2016-06-14:
Yes, everything works perfectly, and by design (of not wasting more memory than needed):

~~~
>>> def foo(): yield
... 
>>> foo
<generator>
>>> f=foo()
>>> f
<generator object 'foo' at 8e81e960>
~~~

dpgeorge, 2016-06-14:
It could be more memory efficient.  Currently a wrapper object is created around a normal function object to make a generator function.  This wapper could be removed by adding a flag to the standard function object.

But the real question is: is it necessary to be compatible with CPython and make a generator function the same type as a normal function?  Is that equality an implementation detail or language definition?

pfalcon, 2016-06-14:
My answer: no, it's not necessary. Nor it's what @peterhinch asks about: "I needed to detect whether a generator or a generator function was passed as an argument". It's possible to distinguish those in MicroPython, and he himself shows how. He seems to be confused by the fact that both generator function type and generator type have the similar external representation, namely `<class 'generator'>` but there's no point to be confused, as it's clear they're different, and there's no warranty that distinct objects will have different external representation. The lowest MicroPython could go is `<class>` (or `<type>`), so it should be well appreciated that for generator func/generator, there's more than that.

~~~
>>> iter([])
<iterator>
>>> iter({})
<iterator>
~~~

pfalcon, 2016-06-30:
Assuming clarified.

closed 2016-06-30 by pfalcon

peterhinch, 2016-07-01:
Indeed. Thank you.