micropython/micropython #2185: unix: build error

peterhinch, 2016-06-14:
Pyboard version builds correctly. Unix: after running make axtls (which produces many warnings but completes) I get
```
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
MISC Preparing upip as frozen module
../tools/make-frozen.py build/frozen_upip/ > build/_frozen_upip.c
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR not updated
GEN build/genhdr/qstrdefs.generated.h
make: Warning: File `build/' has modification time 5.8 s in the future
CC ../py/mpstate.c
CC ../py/nlrsetjmp.c
... [lines omitted] 
CC ../py/../extmod/moduheapq.c
CC ../py/../extmod/moduhashlib.c
../py/../extmod/moduhashlib.c:136:13: error: ‘MP_QSTR_sha1’ undeclared here (not in a function)
     .name = MP_QSTR_sha1,
             ^
make: *** [build/py/../extmod/moduhashlib.o] Error 1
[adminpete@axolotl]: /mnt/qnap2/data/Projects/MicroPython/micropython/unix
$ 
```

dpgeorge, 2016-06-14:
It's a problem with auto-qstr generation.  Please do `make clean` first in unix/ dir.

pfalcon, 2016-06-30:
There're (and always will be) the cases when something is not rebuild as expected at due to some heavy changes (one obvious example is changes to Makefile itself). make clean is a universal solution for any cases like that, foreseen or unforseen. Good practice is to use make clean at the beginning of each work session. I personally use it much more frequently than that.

closed 2016-06-30 by pfalcon