micropython/micropython #2186: esp8266: ADC always returns 65535

deshipu, 2016-06-14:
The following code (taken from quckstart):

```
>>> from machine import ADC
>>> adc = ADC(0)
>>> adc.read()
```

always returns 66535. One possible reason for that may be that the SDK is set to use the ADC for reading the power voltage, not the ADC pin -- this is a compile-time option.

dpgeorge, 2016-06-14:
This example worked fine for me last time I tested it.  Maybe it's a board-specific issue?

deshipu, 2016-06-14:
I'm sorry, I tested on some other boards, and also did ``flash_erase`` on this board and tried again, and it works.

It seems that this setting is stored somewhere in the flash, and that Micropython simply doesn't touch it, leaving it at what it was set before flashing.

closed 2016-06-14 by deshipu