micropython/micropython #2191: RFC: unix: Link with included libffi copy by default

pfalcon, 2016-06-17:
We recently made a change of *requiring* recursive git clone and building an included dependencies for unix port. I'd like to compensate that by not requiring installing libffi-dev distro package to build uPy. We already have submodule for libffi, the caveat that it developers' source and doesn't include generated configure. So, building it requires autotools installed, which is even more stuff to install than just libffi-dev.

So, I'd like to include ready-to build libffi, and the usual question how to do that - using git submodule or pushing directly.

Having an own copy of libffi will also allow to patch out unused stuff we need. See https://github.com/micropython/micropython/commit/c9598604c6872d87b6c818c92efc4bce357d988c for motion is that direction (that was possible to do on upy side, but patching out SSE calling conventions and other niche stuff would need to be done on libffi level). Note that I also envision usage of libffi for baremetal ports.

stinos, 2016-06-18:
> Having an own copy of libffi

Since this is RFC here's the obligatory 'C' when it comes to submodules vs copying source, but I'm pretty sure you know the principle already: yes it allows patching out unused stuff, but it also allows derailing from the original project which is still quite actively developped and so bugs might get fixed upstream which would then require extra effort to get into uPy? Or, if that effort is not done, leave uPy with an out-of-date libffi. Unless of course you plan to make the patching out of unused stuff automatic.

I was also wondering in case you make a copy, would it make sense to fork libffi into uPy's github, patch as needed etc, and then use *that* libffi repo as a submodule? (for benefits like smaller uPy core if ffi is left out, not polluting uPy history with rather unrelated libffi stuff, easier for other projects to reuse)

pfalcon, 2016-06-18:
> derailing from the original project which is still quite actively developed

This is of course the argument for a submodule, and while submodules are chosen for majority of other stuff.

> would it make sense to fork libffi into uPy's github, patch as needed etc, and then use that libffi repo as a submodule?

Sure, that's the plan. (Except that I can't create new repos under micropython account, so create them under mine.)

dpgeorge, 2016-06-26:
Forking libffi, patching and including it as a submodule sounds good to me.

jhdulaney, 2016-07-21:
This is an issue from a distro standpoint, where shipping bundled libraries is discouraged.  You wind up bringing more work on yourselves shipping a forked version of this library, and you introduce the possibility of bugs going unfixed and even security issues.  I fail to see how these disadvantages outweigh the perceived gains here.

jhdulaney, 2016-07-21:
Okay, well, if it is merely the default, then ensure that your make can be passed an option to use external libffi and completely ignore the bundled one.

jonnor, 2024-07-19:
No motion on this over the last 8 years. And this does not seem to be blocking anyone from using MicroPython at the moment. Unless someone steps up to create a merge request, I suggest this can be closed.

closed 2024-07-20 by dpgeorge