micropython/micropython #2194: esp8266: machine reset constants mismatch between docs and 1.8.1 fw

adamwolf, 2016-06-22:
Hi folks!

I am playing with the 1.8.1 binaries sent out to Kickstarter backers, and I noticed a mismatch in the `machine` reset causes between the docs and what's provided in `machine`.

`dir(machine)` shows `'PWR_ON_RESET'`, `'HARD_RESET'`, `'DEEPSLEEP_RESET'`, while the docs have other options like WDT_RESET.

If what's in `machine` is right, and there's an issue in the docs that can be solved by simply removing them, let me know and I can make a PR.

Thanks for all your hard work!

pfalcon, 2016-06-27:
Please be sure to read introduction paragraphs at http://docs.micropython.org/en/latest/esp8266/library/index.html .

pfalcon, 2016-09-04:
Now all five reset causes were implemented.

@dpgeorge , can you check power-on value on Huzzah? We currently have both PWR_ON_RESET and HARD_RESET defined to SDK's REASON_EXT_SYS_RST (6), but the above report suggests that REASON_DEFAULT_RST (0) may be a better match for PWR_ON_RESET. On a NodeMCU-like board, I get REASON_EXT_SYS_RST for both initial power up and hard reset (well, there's no initial power up, as chip reset is controlled by RS232 circuitry).

dpgeorge, 2016-09-05:
@pfalcon On Feather Huzzah I consistenly get 6 for reset_cause().  This is for all of the following cases:
- erase_flash, followed by deploy, followed by boot, followed by pressing reset button (to get a REPL)
- deploy over existing uPy image (so no initial startup needed), followed by pressing reset button (to get a REPL)
- already deployed, just plug in to USB and get a REPL

I never get 0 from reset_cause().

pfalcon, 2016-09-06:
Ok, next idea is that the original report was not about Feather, but about raw Huzzah, which requires external UARt connection. I made a similar setup, and indeed, on power up, .reset_cause() returns 0. So, 742d8bdbe46274401aa261881d14dee50a7618d5.

dpgeorge, 2016-09-07:
> I made a similar setup, and indeed, on power up, .reset_cause() returns 0.

Ok that looks good.

Final issue: docs, cc3200 and stmhal all use POWER_ON, while esp8266 uses PWR_ON_RESET... what to do?

dpgeorge, 2016-09-07:
> Final issue: docs, cc3200 and stmhal all use POWER_ON, while esp8266 uses PWR_ON_RESET... what to do?

Looking at the git history it seems that I added the PWR_ON_RESET constant, and I can't remember why I added that and not POWER_ON, probably just a typo.  So I guess it should be changed to POWER_ON.  Unless we want to change everything (smhal, cc3200) to POWER_ON_RESET so that the constants all end in _RESET.  @pfalcon ?

pfalcon, 2016-09-07:
So, I definitely checked WDT_RESET in docs, etc. when added it, but in a hurry I probably didn't check this one. Well, `machine.POWER_ON` doesn't tell me much what it is. If it's return value from .reset_cause(), it indeed should mention that, and be consistent with the rest of the enumeration. So, I suggest to use PWR_ON_RESET consistently, or POWER_ON_RESET if you want to make it longer.

dpgeorge, 2016-09-08:
I went for PWRON_RESET: it ends in RESET, is short, understandable, and when you read it you parse it as power-on reset, instead of power on reset.  See f3b5480be7243684e31e4631e3da49c725fa7234.

closed 2016-09-08 by dpgeorge