micropython/micropython #2195: MemoryError for trivial code

kentindell, 2016-06-22:
A trivial program, run from REPL on the Pyboard (v1.1), gives a MemoryError. Example:

    >>> def f():
            for i in range(10000):
                s = str(i)
                print(s)

Calling f() from REPL it typically blows up after about five or six thousand times around the loop:

    6319
    6320
    6321
    6322
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "<stdin>", line 3, in f
    MemoryError:
    >>>

Sometimes it doesn't but by re-invoking f() from REPL it usually blows up within a run or two. If I run mem_info() after it explodes I get:

    stack: 432 out of 15360
    GC: total: 102400, used: 1104, free: 101296
     No. of 1-blocks: 21, 2-blocks: 8, max blk sz: 10
    GC memory layout; from 20003000:
    03000: hthhhhhthhBhttth..hthtBht.hthththttttttttt.h...............httt.
    03400: ............hhhhhhhhhhh.httttttthttttt..........................
           (98 lines all free)

Once I've run mem_info() I can't seem to consistently provoke the error again.

I've tried this on versions 1.8.1 and 1.7 (built from source on Ubuntu) with the same issue.

I assume this isn't supposed to happen? It's obviously not very good for performance to create an object then throw it away and let the garbage collector pick up the mess, but this is quite a common style for simple bits of code.

dpgeorge, 2016-06-24:
I saw a similar issue a while ago on the microbit (which has a very small heap, 10k) and can confirm that the above function f() gives a MemoryError on the 32-bit version of unix uPy, with a 10k heap.  @kentindell thanks for finding a reproducible example!

The problem is that stray pointers are remaining at the end of GC blocks preventing the GC from freeing the memory.  In particular, in this case there are 2 blocks being allocated per string: the object structure (mp_obj_str_t) and the data for the string (a short string takes just 1 block on the heap).  On the first round of filling up memory everything is ok and the garbage collection works fine.  But then the second time round filling up memory it just so happens that a block that was used for mp_obj_str_t is reused for the data, and a previous data block is reused for mp_obj_str_t.  The last element in mp_obj_str_t is the pointer to the data.  Because the allocation of the data does not clear the block, and because the string is short (less than 12 bytes), the pointer remains (to the old data, which is now an mp_obj_str_t).  So on the second time round filling up the heap you get an accidental chain of strings whose data (which should have been cleared) references the previous string object.  This prevents garbage collection.

The simple, naive solution is to clear the memory in vstr_init, ie this fixes the current example:
```diff
diff --git a/py/vstr.c b/py/vstr.c
index 3367ae5..1c75226 100644
--- a/py/vstr.c
+++ b/py/vstr.c
@@ -43,7 +43,7 @@ void vstr_init(vstr_t *vstr, size_t alloc) {
     }
     vstr->alloc = alloc;
     vstr->len = 0;
-    vstr->buf = m_new(char, vstr->alloc);
+    vstr->buf = m_new0(char, vstr->alloc);
     if (vstr->buf == NULL) {
         vstr->had_error = true;
         return;
```

But this doesn't fix other cases of this type of bug, and there are many.  When debugging the microbit case I found (through extensive analysis) that there were many (10+) places that needed patching and gave up trying to fix each individually, instead just replacing all malloc's with a call to malloc0.

pfalcon, 2016-06-24:
@dpgeorge : Apparently, what this malloc0 should do is to clear _last 12 bytes_ (for a case of 16 bytes GC block) of the _last_ allocated block.

dpgeorge, 2016-06-24:
> Apparently, what this malloc0 should do is to clear last 12 bytes (for a case of 16 bytes GC block) of the last allocated block.

I don't see how this helps fix the general case of calling code not properly zeroing out the unused parts of newly allocated blocks.  There are lots of cases to fix (not just the one in this issue).

pfalcon, 2016-06-24:
> calling code not properly zeroing out the unused parts of newly allocated blocks

The really serious issue which "calling code" can't deal with is garbage between object end and physical GC block boundary end. Client code isn't aware of it, so it should be handled on system alloc level. Anything else would be a bug in a particular object init code. I'm not aware of such cases (partly because few such bugs were already fixed, and I'm for example keep this issue in mind and write code to take care of that).

dpgeorge, 2016-06-24:
> Anything else would be a bug in a particular object init code. I'm not aware of such cases

I think there are a lot of these cases (based on analysis of writing a pattern to newly allocated blocks, and checking if that pattern still exists later on).  For example mp_obj_exception_make_new does not init/clear traceback_len or traceback_alloc members (which occupy the second word of the block).

dpgeorge, 2016-06-24:
Another few examples of uninitialised heap data:

- code_state.old_globals is never initialised for a generator object (in function gen_wrap_call)
- when creating new qstr pools, the newly allocated but unused data is not zeroed

An argument for zero initialising everything immediately upon allocation is the following: in some cases newly allocated data is not initialised until "later", and another independent allocation can occur before initialisation of the first block is finished.  In this case it could happen that a garbage collection occurs during the second allocation and the GC can't collect some block because it is (wrongly) pointed to by the first allocated block (eg, this can happen in reversed_make_new where initialising o->cur_index can first execute arbitrary code by calling mp_obj_len).

dpgeorge, 2016-06-25:
Other places where memory is sometimes never initialised is in 3rd party libraries, eg sha256 code where sha256_init does not clear the CRYAL_SHA256_CTX.data array.

The most robust way to fix all these problems is to simply make all malloc/realloc calls clear the newly allocated memory before returning.  This is also the most efficient in terms of code size (code size will reduce because we can remove a few memsets).  And after all, the need to initialise the memory (or zero it out) is specific to the GC implementation so the solution should be contained there.  If the GC implementation is replaced by something else (eg scoped heaps) then zeroing the memory is not needed and so the client code (client to the memory manager) should not do it.

kentindell, 2016-08-18:
Just to confirm how the GC sweep works: it determines if there's a pointer to an object by basically grepping the allocated binary blocks to look for words look like pointers and match allocated items and thus arbitrary (and stale) data kicking around in the application can trigger false positives and leave garbage allocated. Is that a fair summary?

I've just been looking through the GC code. If I understand this discussion right the solution is to in a couple of places in `gc.c` change:

    memset((byte*)ptr_in + n_bytes, 0, new_blocks * BYTES_PER_BLOCK - n_bytes);

to:

    memset((byte*)ptr_in, 0, new_blocks * BYTES_PER_BLOCK);

Is that right?

dpgeorge, 2016-08-25:
> Just to confirm how the GC sweep works ... Is that a fair summary?

Yes.

> I've just been looking through the GC code. If I understand this discussion right the solution is to in a couple of places in gc.c change:

Yes, that will zero out all memory.  As I discuss above I think this is necessary because lots of client code (client to malloc) doesn't do this properly.

pfalcon, 2016-08-25:
@dpgeorge : So, as there're no resources to investigate a better way to deal with this, let's add as a stop-gap measure a configurable option for malloc() to behave like malloc0() (for realloc too).

dpgeorge, 2016-08-26:
I made a fix in 5ffe1d8dc07930818bbac6a88bec2aa5bd973402.  It zeros all memory allocated by gc_alloc and gc_realloc.  It does the zeroing in those functions because 1) it was a natural place to do it since the code already zeros part of the block; 2) the m_realloc function in malloc.c does not always know how much existing memory there was in the block that it should not zero out (contrary to gc_realloc, which does know); 3) because of m_malloc_maybe and m_malloc_with_finaliser, there are more places in malloc.c to patch than gc.c.

I tested performance using pystone, and both unix and stmhal ports show a drop in about 2% for the pystone score.  Code size is also reduced a tiny bit with the patch enabled.

The original problem of this issue should now be fixed.

closed 2016-08-26 by dpgeorge