micropython/micropython #2198: Not able to send Multicast messages from ESP8266

mostley, 2016-06-25:
Simple multicast message example that doesn't work on the ESP8266:

```python
import socket

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
data = "test data".encode('utf-8')
sock.sendto(data, ('239.0.0.10', 3535))
```

works with micropython on osx though.

I was experimenting with an Nodemcu v2 ESP8266
MicroPython v1.8.1-31-g9de5eb2 on 2016-06-10

dpgeorge, 2016-06-25:
What is the exact error?  Note that DGRAM is unreliable by design and you may need to send multiple times before it gets through.

mostley, 2016-06-25:
No Exception, result is "9" though. 
I say it's not working because I can't get any messages through. like I said I compared it with a micropython build on osx.
Test setup is as follows:
android phone with the multicast tester app listening for messaged.
osx micropython build as control test
and micropython on nodemcu esp8266 (webrepl or boot.py)

dpgeorge, 2016-06-25:
> No Exception, result is "9" though.

That means it's sending the data.  If the message doesn't get through then maybe the link is not reliable.  Try sending it multiple times in a loop.

The following test works for me:
```py
# SERVER ON *NIX
import socket
s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
print(s.bind(('', 3535)))
while True:
    print("waiting for data...")
    data, addr = s.recvfrom(100)
    print(data, addr)
    print(s.sendto(b'some data back', addr))
```
```py
# CLIENT ON ESP8266
import socket
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
data = "test data".encode('utf-8')
sock.sendto(data, ('192.168.0.9', 3535))
sock.recvfrom(20)
```

mostley, 2016-06-25:
Well yeah, what you are doing here is not a multicast message.
You are addressing the server directly here. 
The multicast address range is defined between 224.0.0.0 and 239.255.255.255 sedd https://en.wikipedia.org/wiki/Multicast_address

dpgeorge, 2016-06-25:
Multicast also works for me, and it is relatively reliable.  Code is:
```py
# RECEIVER ON PC (see http://stackoverflow.com/questions/603852/multicast-in-python)
import socket
import struct

MCAST_GRP = '239.0.0.10'
MCAST_PORT = 3535

s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
s.bind((MCAST_GRP, MCAST_PORT))
mreq = struct.pack("4sl", socket.inet_aton(MCAST_GRP), socket.INADDR_ANY)
s.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, mreq)

try:
    while True:
        print("waiting for data...")
        data, addr = s.recvfrom(100)
        print(data, addr)
except KeyboardInterrupt:
    pass
s.close()
print('done')
```
```py
# SENDER ON ESP8266
import socket
s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
s.sendto(b'some data', ('239.0.0.10', 3535))
```

mostley, 2016-06-25:
what ESP8266 chip and micropython version are you using? your code doesn't work for me either.
I can receive multicast messages with the python server (from app and local) and send messages with the micropython code from osx which are received by the python server and the app.
Only messages send from esp (with either code) aren't received anywhere. which is really weird, everything is on the same wifi, no difference between the clients, no difference in the code. I guess my micropython build has an issue...

mostley, 2016-06-25:
I just tried the latest version (MicroPython v1.8.1-84-g0fb7a7a on 2016-06-25) without success either

dpgeorge, 2016-06-25:
I just built it then and tested it and it worked, MicroPython v1.8.1-84-g0fb7a7a on 2016-06-25; ESP module with ESP8266.  I'm using an ESP12 module (Adafruit HUZZAH feather).

Did you update to the latest esp-open-sdk?  I'm using 03f5e898a059451ec5f3de30e7feff30455f7cec (which uses Espressif SDK 1.5.4).

It might also have something to do with your local network topology.  The TTL is especially important in multicast and you may have too long a path that the UDP packet dies before it reaches the target (and that's considered "successful delivery").

mostley, 2016-06-25:
I'm now on 03f5e898a059451ec5f3de30e7feff30455f7cec too without change in behaviour sadly.
The local topology should be proven since I can send multicast with various other clients as long as the other endpoint isn't the esp8266.
TTL is a good idea though, is there a way to configure that in the esp8266 branch of micropython? I didn't find anything in the docs or the obvious places in the api.

dpgeorge, 2016-06-27:
The TTL defaults to 255 so changing it probably won't make a difference.

mjkillough, 2016-09-24:
I'm also finding the code in [this comment](https://github.com/micropython/micropython/issues/2198#issuecomment-228567062) doesn't work. Sending non-multicast packets from the ESP8266 seems to work fine, but multicast packets get lost without any error. 

I've got a Adafruit Feather HUZZAH with a precompiled version of Micropython that I downloaded from the main site: `esp8266-20160909-v1.8.4.bin`. I'm just installing `esp-open-sdk`, so I'll try and debug what's going on.

mjkillough, 2016-09-24:
It is still happening for me with:
* micropython/micropython@67d52d8cb9f7894a8d50251a67ece3c7a6178270
* pfalcon/esp-open-sdk@d18f815c8d8f6774de4a35652d7305c49f4d1964

To try and rule out my local topology, I compiled this code with the same version of `esp-open-sdk`:

```c
ip_addr_t dest;
IP4_ADDR(&dest, 239, 0, 0, 10);

char *msg = "Hello from ESP";
struct pbuf *buffer = pbuf_alloc(PBUF_TRANSPORT, strlen(msg), PBUF_POOL);
memcpy(buffer->payload, msg, strlen(msg));

struct udp_pcb *pcb = udp_new();
udp_bind(pcb, IP_ADDR_ANY, 3535);

// I have no idea if it is necessary to define the recv callback for this example...
// I did it to be consistent with Micropython's modlwip.c
udp_recv(pcb, udp_recv_cb, NULL);

udp_sendto(pcb, buffer, &dest, 3535);
udp_remove(pcb);
pbuf_free(buffer);
```

This C version sent the multicast UDP message, but might have been dropping some packets - it was hard to tell. The Micropython version never sent the UDP message successfully.

(I can provide the full program if desired - I just hacked it into an example program I found somewhere on GitHub).

mjkillough, 2016-09-25:
Setting `IP_DEBUG` to `LWIP_DBG_ON`, with some other tweaks to LWIP config, allowed me to see what packets were being sent:

```
>>> import socket
>>> s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
>>> s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
>>> s.bind(('', 1900))
>>> s.sendto(b'some unicast data', ('192.168.1.64', 1900))
ip_output_if: ew0
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |        45     | (v, hl, tos, len)
+-------------------------------+
|       12      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |   17  |    0x37d2     | (ttl, proto, chksum)
+-------------------------------+
|  192  |  168  |    1  |   81  | (src)
+-------------------------------+
|  192  |  168  |    1  |   64  | (dest)
+-------------------------------+
netif->output()
17
>>> s.sendto(b'some multicast data', ('239.255.255.250', 1900))
ip_output_if: ew1
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |        47     | (v, hl, tos, len)
+-------------------------------+
|       13      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |   17  |    0x070d     | (ttl, proto, chksum)
+-------------------------------+
|  192  |  168  |    4  |    1  | (src)
+-------------------------------+
|  239  |  255  |  255  |  250  | (dest)
+-------------------------------+
netif->output()
19
```

It seems that this code in `lwip/core/udp.c` is causing it to use a different `netif` (with different `src` address) when sending multicast packets:

```c
  /* find the outgoing network interface for this packet */
#if LWIP_IGMP
  netif = ip_route((ip_addr_ismulticast(dst_ip))?(&(pcb->multicast_ip)):(dst_ip));
#else
  netif = ip_route(dst_ip);
#endif /* LWIP_IGMP */
```

For me `pcb->multicast_ip==0`, which I _think_ is causing `ip_route()` to return `netif_default`. I think `192.168.4.1` is the IP the chip has when it boots up in AP mode? (I've since configured it to join my home network in STA mode, where it gets given `192.168.1.81` over DHCP).

I am not sure what the correct solution is. [`espconn_udp.c`](https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/src/app/espconn_udp.c) seems to play around with `netif_set_default()` when sending UDP packets. [Another example](https://github.com/willemwouters/ESP8266/wiki/UDP-Server---LWIP-Stack) fiddles `pcb->multicast_ip` itself. Upstream LWIP [suggest it should be set via `udp_set_multicast_netif_addr()`](http://savannah.nongnu.org/bugs/?38061).

Separately, it looks like it can be set via `lwip_setsockopt()`, along with a bunch of other useful options which would allow us to receive multicast UDP packets using `IP_ADD_MEMBERSHIP`. Is there a reason that's not plumbed through `modlwip.c`? I'd be happy to give that a go.

dpgeorge, 2016-09-27:
@mjkillough is pcb->multicast_ip ever initialised, or is 0 a default value by accident?

In modlwip netif_set_default() is called when you construct an interface, so the default interface will be the last one constructed.  Try deactiviting if1, then constructing if0, so that if0 (STA) is the default.

mjkillough, 2016-09-27:
> is pcb->multicast_ip ever initialised, or is 0 a default value by accident?

It looks like it gets initialized to 0 when `udp_new()` sets the entire `struct udp_pcb` to 0. As far as I can see, LWIP will only set it if `setsockopt()` is called, which Micropython never calls.

> In modlwip netif_set_default() is called when you construct an interface, so the default interface will be the last one constructed. Try deactiviting if1, then constructing if0, so that if0 (STA) is the default.

I can only see `modlwip.c` call `netif_set_default()` for `lwip_slip_make_new`. For the non-slip case, I can't see where it is called?

I'll have another look at what happens in `ip_route()` to see which interfaces lwip thinks are active at that point. (Is there a way to build Micropython with gdb? I'm currently doing a lot of `printf` debugging).

I chatted about this issue on IRC last night with eldios_lele, and we found:
- Sending multicast UDP packets doesn't seem to work on his NodeMCU.
- (Perhaps unsurprisingly) it does work for me esp8266 when I connect my laptop directly to the esp's AP. In that case, the esp's IP matches the one it puts in the UDP src field for multicast packets.

mjkillough, 2016-09-27:
> Try deactiviting if1, then constructing if0, so that if0 (STA) is the default.

Ah, I see! I misunderstood this at first. Now that my brain is engaged, I see you mean to do:

```python
import network
ap = network.WLAN(network.AP_IF)
ap.active(False)
```

... which does indeed cause the STA interface to become the default, allowing multicast to work. Thanks!

I hadn't realised that the AP interface continued to be active after I did `network.WLAN(network.STA_IF).active(True)`. (I see now that the docs do say, "You can then disable the access-point interface if you no longer need it", so perhaps I needed to read the manual...)

If it is possible for both interfaces to be active at the same time, then it seems reasonable enough for it to send multicast packets to the AP interface by default. However, when bother interfaces are active, it seems AP is always the default interface, no matter which combination of `.active(True)`/`.active(False)` I do.

Unless I'm being daft, perhaps `modnetwork.c` could be extended to allow the user to specify which interface they want to be the default?

dpgeorge, 2016-09-28:
> perhaps modnetwork.c could be extended to allow the user to specify which interface they want to be the default? Or perhaps socket.bind(('ip', port)) could set udp_pcb->multicast_ip?

I'm not sure what the "correct" thing to do is, but exposing the relevant config via the Python setsockopt method seems the most general.  @pfalcon what do you think?

EternityForest, 2017-03-30:
I tried mjkillough's idea of having bind() set the multicast_ip, and I'm now able to have 2 sockets open, one for sta and one for ap mode, and I can send multicast frames from each and they work.

I don't know what the correct way to handle this either, but having multicast packets sent from the same address you bind to makes sense to me. Is there any case where you would want to bind to one interface but send multicasts from another?

I wasn't able to get udp_set_multicast_netif_addr to compile so I just found the definition of that macro and copy/pasted it.

In case anyone else is interested, Here's the relevant change in lwip_socket_bind
```
          case MOD_NETWORK_SOCK_DGRAM: {
              err = udp_bind(socket->pcb.udp, &bind_addr, port);
++            ip_addr_copy(socket->pcb.udp->multicast_ip, bind_addr);
              break;
          }
```

mjkillough, 2017-04-06:
> Is there any case where you would want to bind to one interface but send multicasts from another?

If there is such a case, I think it would be reasonable to require that a socket be opened on each interface.

nickovs, 2018-08-12:
If the tweak suggested by @EternityForest works then I suspect the correct solution is for the `IP_MULTICAST_IF` socket option to be implemented and have it set `socket->pcb.udp->multicast_ip`. I don't know when I'll have the free time to look into this (probably not until late September).

UltraPhil, 2021-08-11:
Hi,

I'm currently running into multicast problems as well, on my ESP32 with MicroPython v1.16 (2021-06-23).
The ESP32 is connected to my Wi-Fi network (WLAN in STA_IF mode) and the built-in AP is disabled.

Here is my test code:
```python
MCAST_GRP = '225.128.64.32'
MCAST_PORT = 49150
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)
sock.sendto(some_data, (MCAST_GRP, MCAST_PORT))
```
While running above code, I sniffed the traffic with WireShark. The only multicast traffic getting out of the ESP32 is mDNS-related (which seems to happen when the wlan interface reconfigures itself). But I see no multicast packets whatsoever.

The same code, on vanilla python, produces the expected result.

I'm not sure I understand what is meant by @EternityForest tweak. I tried to bind the socket to `network.WLAN(STA_IF).ifconfig()[0]`, which is the interface IP address, but it doesn't work either.

Have things much changed since v1.8? I am doing something wrong?

Thanks a lot!