micropython/micropython #2202: CI support for esp8266 port

pfalcon, 2016-06-27:
I leisurely worked on this last days, and after a lot of trial and error, here's something which works: https://github.com/pfalcon/micropython/commits/travis-esp8266 . It's not integratable into the current main .travis.yml, so I suggest we just push that branch to the main repo and updated from time to time, as few other similar branches. I also suggest we adopt travis-* naming pattern for such CI branches.

dpgeorge, 2016-06-29:
> It's not integratable into the current main .travis.yml

I guess that's because 1) it uses sudo:false, and 2) it can take ages to build the toolchain if not in the cache already.

> I suggest we just push that branch to the main repo and updated from time to time,

Yes that sounds sensible.

> I also suggest we adopt travis-* naming pattern for such CI branches.

There is already the travis-testing branch.

pfalcon, 2016-06-29:
>> It's not integratable into the current main .travis.yml

> I guess that's because 1) it uses sudo:false,

It uses sudo: false, just as esp-open-rtos, because literally the same travis.yml fails to build the toolchain with sudo: true - it errors out building gdb on Python dependency, even though all packages are installed. So yes, can't use sudo: true, and converting our travis.yml to sudo: false would take another bunch of effort.

> and 2) it can take ages to build the toolchain if not in the cache already.

Actually it's 20-something minutes. Note that there's currently no automatic cache reset, it needs to be cleaned manually (from Travis UI) on esp-open-sdk update.

> There is already the travis-testing branch.

And travis-clang too (I should have left a comment to related commit).

pfalcon, 2016-06-30:
Build is green at https://travis-ci.org/micropython/micropython/builds/141380253 .

closed 2016-06-30 by pfalcon

nevercast, 2019-06-25:
@pfalcon @dpgeorge Is this still current? Does this apply to the ESP32 port also? We have CI for our ESP32 firmware that auto-builds whenever micropython/micropython#master is pushed to, I can look in to CI for ESP32 if this is not something micropython already do.

dpgeorge, 2019-06-25:
@nevercast please see #4867, which I'm working on right now.  It already includes esp8266 builds (takes 1-2mins) and esp32 is WIP.