micropython/micropython #2206: httpget inside a Timer freezes a ESP8266

MichaelSal, 2016-06-29:
I try to put an httpget like in the basic guide inside a Timer, but the entire ESP8266 system freezes.
```
def http_get(url):
    _, _, host, path = url.split('/', 3)
    addr = socket.getaddrinfo(host, 80)[0][-1]
    s = socket.socket()
    s.connect(addr)
    s.send(bytes('GET /%s HTTP/1.0\r\nHost: %s\r\n\r\n' % (path, host), 'utf8'))
    while True:
        data = s.recv(100)
        if data:
            print(str(data, 'utf8'), end='')
        else:
            break
```
```
import socket
from machine import Timer
 tim = Timer(-1)
 tim.init(period=1000, mode=Timer.ONE_SHOT, callback=lambda t:http_get("http://google.com/"))
```

dpgeorge, 2016-06-30:
Issue confirmed.  It's locking up in the `data = s.recv(100)` call.  Probably because it's calling ets_loop_iter (for the recv) from within ets_loop_iter (which is handling the timer callback).

doudz, 2016-08-05:
I have a slightly different bug, I'm using http_get in a periodic Timer every 60sec and it's almost working except that the first http_get does nothing, the second does the first one, the third does the second , etc etc
it looks like http_get is locked somewhere waiting for the next Timer callback to be really execute.

In my case it's not a big issue since I'm doing temperature logging but I'm loosing accuracy.

pfalcon, 2016-08-09:
So, the current suggestion is to treat esp8266 timers as hardware timers, and avoid doing complex operations in their callbacks.

stlk, 2016-12-05:
Are there any news or workarounds? Just stumbled upon this issue with `esp8266-20161110-v1.8.6.bin`. Interesting fact is that MQTT works inside the timer callback.

dpgeorge, 2016-12-07:
> Are there any news or workarounds?

There is no progress on this issue.  To work around it you can set a flag in the timer callback, and then check this flag in your main loop and act accordingly.

dpgeorge, 2017-03-20:
This is now fixed by #2878.

closed 2017-03-20 by dpgeorge