micropython/micropython #2209: mpy-cross creates corrupt .mpy's on windows

ghseb, 2016-06-30:
I try to use frozen bytecode in stmhal in a windows/mingw environment. After compiling a python script to micropython bytecode with mpy-cross.exe i get a .mpy file as expected. If i then apply the mpy-tool.py on the .mpy file it raises an IndexError:

```
Creating build-STM32L476DISC/frozen_mpy.c
python3.5 ../tools/mpy-tool.py -f -q build-STM32L476DISC/genhdr/qstrdefs.preprocessed.h build-STM32L476DISC/freeze/test_script.mpy > build-STM32L476DISC/frozen_mpy.c
Traceback (most recent call last):
...
File "../tools/mpy-tool.py", line 357, in read_uint
b = bytes_cons(f.read(1))[0]
IndexError: index out of range
Makefile:301: recipe for target 'build-STM32L476DISC/frozen_mpy.c' failed
mingw32-make: *** [build-STM32L476DISC/frozen_mpy.c] Error 1
mingw32-make: *** Deleting file 'build-STM32L476DISC/frozen_mpy.c'
```

When i compared the bytecode that is generated in linux environments to the bytecode in windows environment the bytecode differs when it comes to a newline character in the bytecode (0xa). It seems that when the bytecode is generated by mpy-cross it prints out a newline character somewhere and this is resolved to a linefeed-only in linux and to a linefeed + carriage return in windows. If the additional carriage return is removed freezing works.

I think the additional carriage return is not comming from the input python scripts, because in my windows environment i already tried linux encoded line endings and mpy-cross nevertheless failed.

Simple python script that produces the error (with windows newlines, apart from others that i tried):
[test_script_class.py.txt](https://github.com/micropython/micropython/files/341608/test_script_class.py.txt)

Resulting .mpy on windows
[test_script_class_win.mpy.txt](https://github.com/micropython/micropython/files/341586/test_script_class_win.mpy.txt)

Resulting .mpy on linux
[test_script_class_linux.mpy.txt](https://github.com/micropython/micropython/files/341588/test_script_class_linux.mpy.txt)

Until now i could not figure out where the problem comes from.

br,
Sebastian

pfalcon, 2016-06-30:
My suggestion would be just to use https://msdn.microsoft.com/en-us/library/ktss1a9b.aspx (_set_fmode()) and don't bother with non-compliant O_BINARY/O_TEXT. This issue likely affects micropython executable itself too. @stinos: ?

ghseb, 2016-07-01:
I can confirm that your suggestion fixes the issue if i compile mpy-cross with mode `_O_BINARY` set. But since im in a mingw environment which has no `_set_fmode()` i had to set the global variable `_fmode` directly to `_O_BINARY`.

pfalcon, 2016-07-17:
@stinos : Ping again for your opinion.

stinos, 2016-07-19:
@pfalcon sorry I was away from computers for a while. Will check this tomorrow.

stinos, 2016-07-20:
Since ```fopen``` is banned from uPy (unless I'm mistaken) using ```_set_fmode``` or similar is indeed the way to go. At least I don't know another way to get rid of endline translation. To test I put it in ```windows/init.c``` and all tests still pass which makes sense because the test runner does it's best to make everything newline-agnostic so the comparision with the CPython output never fails.
It does affect uPy in 2 ways though:
- not CPython-compliant when it comes to newlines as it always writes ```\n```. But that is actually just the other way around compared to how it is now: uPy always writes ```\r\n``` no matter if the open mode for python's ```open``` is specified as ```b``` or not, which also doesn't comply. I don't think this really is an issue.
- more important: reading text files which use ```\r\n``` will preserve the ```\r```. This might screw up existing programs but it's hard to tell to which extent it will have an influence. This could be fixed by e.g. switching fmode depending on ```b``` being passed to python code or not, Or by ignoring any ```\r``` characters in one of the stream read functions, but none of those matter for this issue. So we could just add it to the CPython differences doc and have users deal with it themselves - I wouldn't mind that.

Conclusion: go with @pfalcon's suggestion for now, make sure behaviour is ducumented, and if really ever needed someone can come up with another fix for dealing with line-endings on windows later.

pfalcon, 2016-07-20:
To clarify, I'm looking for an ack for the following changes:

1. Decide hot to fix this for mpy-cross. Based on @ghseb's note on mingw, setting `_fmode` appears to be the way, but maybe you, @stinos , have any concerns or other suggestions.
2. As the same issue ultimately affects the main micropython executable too, see if it makes sense to apply fix from p.1 to it too. I agree that producing broken binary files is much worse than producing text files with another popular line-ending convention. I assume this point is decided.

stinos, 2016-07-20:
1. Only tried _set_fmode so far and that works as expected, but setting _fmode has for some reason no effect for an msvc build (nor does _set_fmode alter _fmode here). No idea if it's a bug or a build problem or I'm misunderstanding the documentation. Given that I wouldn't set _fmode directly but use _set_fmode and provide an implementation of it for mingw (i.e. a function which just sets _fmode). And then call it in mpy-cross/main.c.
2. I'm +1 on applying same change to main uPy. And if done in windows' `init()` function, it's immediately fixed for mpy-cross as well.

pfalcon, 2016-07-20:
Ok, so would you prepare patch for p.1, tested with msvc/mingw ?

pfalcon, 2016-07-22:
The original issue reported fixed by the commit above. Main micropython still would produce broken binary files, but let's leave that to another bugreport.

closed 2016-07-22 by pfalcon