micropython/micropython #2214: Prepare for 1.8.2 release

pfalcon, 2016-07-02:
General:

* ???

ESP8266:

* https://github.com/micropython/micropython/issues/2208 (would be nice)
* umqtt tested/included
* upysh pending changes
* KS update: Bring up to backers question of providing binary for free download

robert-hh, 2016-07-02:
Hi Paul. SInce Damien's first commit, I had moved the modules from esp8266/scripts to esp8266/modules, and upysh as well. No complaints, but I did not try all of them. The ones I tried as frozen bytecode are webrepl, ntptime and dht (and upysh). I do not have suitable devices for trying the other modules.

pfalcon, 2016-07-02:
Thanks for confirming, @robert-hh, I now moved both webrepl and webrepl_setup to frozen bytecode. That should give the big boost, as webrepl is always there (in RAM), even if people don't use it. And should give a good test coverage before switching even more stuff to bytecode...

pfalcon, 2016-07-08:
Ok, per the list above, 1.8.2 should ready for tagging.

dpgeorge, 2016-07-10:
Done!  See 1459a8d5c9b29c78da2cf5c7cf3c37ab03b34b8e.

closed 2016-07-10 by dpgeorge