micropython/micropython #2216: SIGPIPE handling problem

pfalcon, 2016-07-03:
Here's another quirk of POSIX: writing to a file descriptor may lead to process receiving SIGPIPE, the default action is silent termination. This is the expected and natural behavior in case of pipe. However, the same behavior applies to a socket too. So, you run an app, and it seemingly exits half-way thru without any error message. Cute. And because this behavior is actually useful in case of pipe (micropython foo.py | less), we can't just disable SIGPIPE globally.

So, need to think what to do about it. First step is probably to figure out how often sockets are affected. I got it when writing to non-connected socket. Peer disconnect should not lead to EPIPE delivered as SIGPIPE, but to ECONNRESET instead, at least that's what I've been seeing, though internets discuss cases when people had SIGPIPE issue in such cases. But even with writing to non-connected socket, it's pretty unpleasant - that wouldn't happen often, the more time and pulled hair it would take to figure it out from a non-suspecting user.

pfalcon, 2016-07-06:
Some info: http://stackoverflow.com/questions/33053507/econnreset-in-send-linux-c

pfalcon, 2016-11-02:
In a standard echo effect, other people now being hit by this too, we got 2 reports in last week.

dpgeorge, 2016-11-02:
It's a bit similar to the EINTR problem.  How does CPython handle SIGPIPE?

pfalcon, 2017-05-01:
CPython handles SIGPIPE in an easy way:

~~~
PyOS_setsig(SIGPIPE, SIG_IGN);
~~~

So, we'll be doing that now too.

pfalcon, 2017-05-01:
11bc21dfa8f087ae2d95f378267ef270a85029ad

closed 2017-05-01 by pfalcon