micropython/micropython #2217: Changing semantics of utime.ticks_diff()

pfalcon, 2016-07-03:
Point 1: usabality/cosmetics

Initially, it seemed pretty natural for that func to have signature utime.ticks_diff(time_before, time_after). However, that function is intended to replace expression time_after - time_before, so it's kinda actually more natural to have it utime.ticks_diff(time_after, time_before).

Point 2: functionality

ticks_diff() is defined to return unsigned number. This allows to use full range of ticks_ms/us values, but misses ability to detect real-world useful condition of "deadline missed". This issue is being raised as a result of trying to convert uasyncio to use ticks_ms(). Lack of ability to properly detect missed deadline is a blocker for such conversion (currently, missed deadline results in a very large time difference, such difference used as a wait delay).

Note re: Point 1: It is NOT too late to fix that issue. But it may become too late at some time, and then may become a regular source of confusion (I judge by myself, again, initially it seemed quite natural to use (time_after, time_before), the problem is that it's not always immediately clear which time is "before", and in real-world cases time_after may be before "time_before" (missed deadline); so, relying on pure mathematical order of subtraction arguments turns out to be a better idea).

pfalcon, 2016-07-11:
@dpgeorge : I'd appreciate a response on the point 2 above, as it blocks uasyncio refactoring.

dpgeorge, 2016-07-11:
I don't think we should change semantics (even though I also sometimes feel the order should be the other way around) because it's already out there in the wild.

Re point 2, what exactly is the proposal to change the behaviour to so it can detect deadline missed?

pfalcon, 2016-07-11:
> because it's already out there in the wild.

Everything we have is at best "provisional" status in CPython terms, and we have explicit warning regarding that: "WARNING: this project is in beta stage and is subject to changes of the code-base, including project-wide name changes and API changes.". So, should not hesitate to change things for better instead of carrying cruft around.

> Re point 2, what exactly is the proposal to change the behaviour to so it can detect deadline missed?

Ah, ok, the proposal is to make return value signed instead of unsigned.

peterhinch, 2016-07-12:
+1 for signed return value

dpgeorge, 2016-07-12:
> Ah, ok, the proposal is to make return value signed instead of unsigned.

Please explain in detail with examples, I don't see how this works and why it's useful.

peterhinch, 2016-07-12:
Yes. And the scheduler will presumably prioritise the most overdue task.

dpgeorge, 2016-07-12:
> How a typical scheduler works is tags each task with an absolute time of next wakeup,

Are you intending to use the return value of ticks_ms() to create an *absolute* time value?  But that's not possible because those return values are supposed to be opaque and only useful as inputs into ticks_diff() (as per the docs).

> Suppose that task is scheduled to run at time 10, and it so happens that current time is 11

How do you compute time "10" and "11"?  What do they mean?  You can't do `ticks_ms() + 2` for example, because it may overflow the range used by those functions (and may become a bignum).

> ticks_diff(11, 10)

How do you intend to re-write ticks_diff so this is a negative number?  The whole point of ticks_diff is that it takes care of wrap-around and the result of the above should be a large number.

pfalcon, 2016-07-12:
> Are you intending to use the return value of ticks_ms() to create an absolute time value? But that's not possible because those return values are supposed to be opaque and only useful as inputs into ticks_diff() (as per the docs).

Ok, so as a next step, ticks_add() will be needed ;-).

> How do you intend to re-write ticks_diff so this is a negative number?

Simply by not converting subtraction result to an unsigned value.

> The whole point of ticks_diff is that it takes care of wrap-around and the result of the above should be a large number.

It will do all the same, except: 1) will be able to return negative difference; 2) will cut effective range 2 times, e.g. if before ticks_diff could produce correct answer if two events were 2^30 time units apart (and incorrect otherwise), now it will produce correct answer if two events are apart 2^29 time units - but apart in both ways, one before another, or vice versa.

That's usual and pretty adequate means to allow to deal with things which don't appear in a predefined order, so I don't propose to add ticks_diff_signed() (especially given that ticks_add will need to be added, etc.)

Even more mind-boggling things to do are in queue, like defining spur-of-the-moment total order on items of a mathematical ring, where there's no strict ordering: https://github.com/micropython/micropython/issues/2219

peterhinch, 2016-07-13:
I've used the following approach (empirical evidence shows that the us timer rolls over at 0x3fffffff i.e. 2 MSB's are zero as expected).
```python
TIMERPERIOD = 0x3fffffff
MAXTIME = TIMERPERIOD // 2

def ticks_diff(before, after):
    diff = (after - before) & TIMERPERIOD
    if diff < MAXTIME:
        return diff
    return diff - TIMERPERIOD - 1
```
This yields
```
>>> ticks_diff(100, 110)
10
>>> ticks_diff(120, 110)
-10
>>> ticks_diff(MAXTIME - 2, MAXTIME + 2)
4
>>> ticks_diff(MAXTIME + 2, MAXTIME - 2)
-4
>>> utime.ticks_diff(MAXTIME - 2, MAXTIME + 2)
4
>>> 
```

dpgeorge, 2016-07-13:
Well, if we are going to substantially redefine things so that addition and "absolute" values are possible, then yes that's a possibility.  One could define that the result of ticks_* functions are numbers (integers) that form a periodic time value, modulo N, and that the user can do arithmetic on them.  Then ticks_norm(t) normalises t to be in the range [0, N).  (That means no need for ticks_add, it's just ticks_norm(a + b)).  But I think you still need ticks_diff (not ticks_norm(a - b)) if you want to extend the range of return values to (-N, N).

peterhinch, 2016-07-13:
Agreed. At risk of stating the obvious addition is necessary to determine a ticks value for future events specified as an offset from now, and a means of creating a ticks value from an absolute time also has application for future events.

Another issue is how to handle out of range results. ticks_norm() presumably accepts any integer value and implies that the caller takes responsibility for overrange. ticks_add and ticks_diff offer the option of raising an exception if you guys think it's the way to go.

pfalcon, 2016-07-13:
> Well, if we are going to substantially redefine things so that addition and "absolute" values are possible, then yes that's a possibility. One could define that the result of ticks_* functions are [...]

When idea of ticks_ms()/ticks_us() was presented on python-ideas, there was question whether we want to provide a module for a general modular arithmetic. The answer was no, and I think it still holds. So, we should change/extend utime API to make it practical for real-world embedded time management, that's it.

> Then ticks_norm(t) normalises t to be in the range [0, N). (That means no need for ticks_add, it's just ticks_norm(a + b)). 

That has a risk of overflowing from short to long int, which ticks_add() precludes.

> But I think you still need ticks_diff (not ticks_norm(a - b)) if you want to extend the range of return values to (-N, N).

To [-N/2-1, N/2] specifically.

dpgeorge, 2016-07-13:
> To [-N/2-1, N/2] specifically.

Then the output of ticks_ms() etc should be in the range [0, N/2) so that ticks_diff always returns a positive number for time-ordered input values.

The alternative to all this is write a scheduler that uses relative times only, not absolute (which is where the problem stems from).

pfalcon, 2016-07-13:
> Then the output of ticks_ms() etc should be in the range [0, N/2) so that ticks_diff always returns a positive number for time-ordered input values.

No, that's not required. It's just ticks_diff gets defined to provide valid result only if its arguments are apart less than N/2 (but in any order).

> The alternative to all this is write a scheduler that uses relative times only, not absolute (which is where the problem stems from).

It's unlikely possible to write an efficient scheduler for such case, and asyncio anyway requires call_at() call which takes absolute time.

dpgeorge, 2016-07-13:
> No, that's not required. It's just ticks_diff gets defined to provide valid result only if its arguments are apart less than N/2 (but in any order).

I think it should be required to return only N/2 values from ticks_ms.  It should be that ticks_diff(t1, t2) should always be positive if t1 was set from ticks_ms() before t2 was set from ticks_ms().  Negative values from ticks_diff() should only be possible if you are passing in args that occured in the wrong time-order.

> and asyncio anyway requires call_at() call which takes absolute time.

Then the whole uasyncio module requires the concept of periodic time (also the asyncio.time() method would return a periodic value), and therefore periodic arithmetic.

pfalcon, 2016-07-13:
> I think it should be required to return only N/2 values from ticks_ms. It should be that ticks_diff(t1, t2) should always be positive if t1 was set from ticks_ms() before t2 was set from ticks_ms().

I don't see immediate need for that. The dimension of ranges for both ticks_ms and ticks_diff is N, it's just the ranges themselves are [-N/2; N/2] and [0; N) respectively. "Cut everything in 2 just because one thing was cut in 2" seems like logical error (with a perspective to be applied recursively). Note that even if ticks_diff value is signed, it can be converted to unsigned if the order of A, B events is known, to get back to full [0, N) duration range (that will require exposing N as a constant).

> Negative values from ticks_diff() should only be possible if you are passing in args that occurred in the wrong time-order.

It's not possible to achieve such requirement because values are, well, periodic. It's ok to cut range in 2 if brings some benefits, but cutting range in 2 everywhere won't change much, except make overflow happens sooner and more often (even for the cases where otherwise local cut-in-2 didn't apply).

pfalcon, 2016-07-13:
> Then the whole uasyncio module requires the concept of periodic time (also the asyncio.time() method would return a periodic value), and therefore periodic arithmetic.

Yes, hence https://github.com/micropython/micropython/issues/2219, and patch referenced from it, which makes heapq properly work with "periodic" time, as long as all events currently in queue (and a new one added) are spread within "same" N/2 ring interval. See http://math.stackexchange.com/questions/1211454/in-mod3-is-3-greater-than-or-less-than-1 for ideas behind that.

dpgeorge, 2016-07-13:
If you are allowing things like `eventloop.call_at(eventloop.time() + 10)` then you need something like ticks_norm.

pfalcon, 2016-07-13:
> If you are allowing things like eventloop.call_at(eventloop.time() + 10) then you need something like ticks_norm.

That all is a big topic, consisting of many subtopics. While it would be nice to make big self-consistent plan in one step, that's not realistically possible (I for one don't have resources to craft such masterplan). What's clear is that this line is a big problem: https://github.com/pfalcon/micropython-lib/blob/uasyncio-ticks_ms/uasyncio.core/uasyncio/core.py#L57 and to fix it, ticks_diff() needs to return signed value. Once that is fixed, can move to the next issue, etc.

dpgeorge, 2016-07-13:
> That all is a big topic, consisting of many subtopics. While it would be nice to make big self-consistent plan in one step, that's not realistically possible

A big plan needs to be made possible.  It (question of time resolution and the user specifying times in asyncio) is really the first thing that needs to be discussed and planned before moving to a specific implementation.  Doing ad-hoc changes (and changing existing APIs) without a grand goal seems like a bad idea to me, and seems like it would require further changes down the line.  For example, maybe we need ticks_add and ticks_sub and ticks_norm.  Or maybe we need these functions to accept big-num so users can do addition.

So I'd suggest openning a discussion (it's started here) on how we want the top-level uasyncio API to look and work when the user needs precision.

pfalcon, 2016-07-13:
There's grand goal - port uasyncio.core to baremetal, no assumptions on non-wrapping clock, no assumptions on floating point availability. No allocation allowed in scheduler, so big-nums are out of question. That all obviously leads to utime.tick_ms(). Yes, APIs are needed to do more extended operations on modular time, I'm trying to adapt existing API step by step, though yes, it's possible to say that ticks_add() and ticks_sub() are needed, and ticks_diff() should be deprecated (like, really deprecated, to be removed from all concise ports, as it's not ok to have different argument order for 2 similar functions - ticks_sub returning signed result, and ticks_diff keeping returning unsigned).

Then the only "risk" is that entire uasyncio main loop will be rewritten in C, and these "partial" changes will be futile. That's not however the case, because our aim is to allow to write real applications in Python, so even if some part eventually gets optimized to C, that's only optimization, not new functionality (i.e. the same functionality should be easily and efficiently implemented in Python too).

I obviously follow the same evolutionary approach, so if there's clear and unavoidable need to do signed subtraction of ticks_ms value, I bring it in. Yes, it's possible to look a step ahead and think of -ticks_diff, +ticks_sub, ticks_add. But that's all so far. There will be further changes down the road, and that's good.

deshipu, 2016-07-13:
Would it make sense to have a general library with utility functions for handling wrapping numbers, ring buffers, running averages and similar things that are quite common in the microcontroller world?

dpgeorge, 2016-09-20:
I think we should move forward on this.  I'm happy to concede that ticks_diff should indeed have its arguments reversed (using it more now I find it confusing the way it is, maybe because this discussion is now always in the back of my mind!).

I think we need to have ticks_add() and ticks_diff(), and possibly also ticks_sub().  Note that addition only has one set of possible args: a ticks opaque value and a meaningful/transparent integer, and the return value is a ticks opaque value.  But subtraction can either be between 2 opaque values (returning a meaningful integer, current behaviour of diff), or between 1 opaque and 1 meaningful, returning opaque.  The latter is really a case of addition, but adding a negative number, ie `ticks_sub(t, i) == ticks_add(t, -i)`.

pfalcon, 2016-09-20:
> I think we should move forward on this. I'm happy to concede that ticks_diff should indeed have its arguments reversed (using it more now I find it confusing the way it is, maybe because this discussion is now always in the back of my mind!).

Wow, great, because my plan for this was: to introduce ticks_sub, and deprecate ticks_diff. But I think we should have better (like, more liberal) policy of *fixing* experimental things (all things not in CPython are "experimental").

> But subtraction can either be between 2 opaque values (returning a meaningful integer, current behaviour of diff), or between 1 opaque and 1 meaningful, returning opaque. 

Yes, that nails it down, but I'm not sure how far we can go with that. Having all of ticks_add, ticks_diff, ticks_sub seems like overkill and would-be source of confusion for many people.

> The latter is really a case of addition, but adding a negative number, ie ticks_sub(t, i) == ticks_add(t, -i).

So, maybe that's the way how it should be handled (i.e. explicitly documented).

peterhinch, 2016-09-20:
+1 for ticks_add() but I'm not sure if we need ticks_sub() **and** ticks_diff(). Why not just have ticks_sub() behaving as described above?

If an undocumented, deprecated ticks_diff() were retained for compatibility it would surely make sense to keep the existing arg order.

dpgeorge, 2016-09-21:
> If an undocumented, deprecated ticks_diff() were retained for compatibility it would surely make sense to keep the existing arg order.

As @pfalcon says, better to fix it now rather than later, or rather than deprecate it.

So then let's go with ticks_diff(t2, t1) and ticks_add(t, i).

pfalcon, 2016-10-29:
> So then let's go with ticks_diff(t2, t1) and ticks_add(t, i).

Ok, this missed previous release, but no time to postpone it more if it gets into 1.8.6 (need to announce it beforehand, propagate changes, etc.)

So: https://github.com/micropython/micropython/commit/6ed5583f8cfa5aa0afb776654cd996abaaa5fba3

I will work on tests and docs tomorrow.

pfalcon, 2016-10-29:
And the funny thing about patch above is that ticks_diff is now actually equivalent of just normal subtraction, i.e. instead of `ticks_diff(a, b)` one can write `a - b` and get correct result. Indeed, with a current definition of ticks_ms/us/cpu, which is "guaranteedly non-negative small int", just applying subtraction will leave it in small int range, just may be negative. @dpgeorge , what do we do about it? ;-)

And I found another use for ticks_add(): ticks_add(0, -1) will return the wraparound value for ticks.

pfalcon, 2016-10-29:
> And the funny thing about patch above is that ticks_diff is now actually equivalent of just normal subtraction, 

Well, it's just incorrectly implemented.

pfalcon, 2016-10-29:
ticks_add implemented in d86cac4b8268bbe9203cdd7ab8d8c2a65d142986

pfalcon, 2016-10-30:
ticks_diff fixed in e429daa5724e5e3c5f8866742abc64eb90ded71d . A test needs cleanup before pushing.

pfalcon, 2016-10-30:
Test pushed in be6a765c69baf6f0bae77379438fb11c8f2c104a

pfalcon, 2016-10-30:
ticks_diff() docs updated in d60ad5cf5356b7600ed53365216405c546fea299 . Note that ticks_ms() and ticks_diff() description is now formally detailed, but perhaps become too scary for some users. That's a usual problem of formal spec vs user docs. Well, let's create market for uPy tutorials ;-).

Type/grammar fixed are as usual welcome. ticks_add() is still todo.

dxxb, 2016-10-31:
Hi @pfalcon, I was wondering if ticks_diff() should throw an exception when the difference cannot be represented by the return value's type. My thinking is: the exception is preferable because it allows the caller to decide how to handle the 'out of range' condition.

pfalcon, 2016-10-31:
> I was wondering if ticks_diff() should throw an exception when the difference cannot be represented by the return value's type. My thinking is: the exception is preferable because it allows the caller to decide how to handle the 'out of range' condition.

No, ticks_* are performance-critical functions which should not do any checks. They are also functionality-critical, and simply cannot fail, they're specifically defined using mathematical apparatus where it's always possible to represent result in return value's type and "out of range" condition is not possible.

dxxb, 2016-10-31:
> No, ticks_* are performance-critical functions which should not do any checks. They are also functionality-critical, and simply cannot fail

ack, if it cannot fail there is no point discussing the decrease in performance.

> they're specifically defined using mathematical apparatus where it's always possible to represent result in return value's type and "out of range" condition is not possible.

Well, it is possible and when it happens it's dealt with in a specific way, as the commit message says:
> if a follows b in more than MICROPY_PY_UTIME_TICKS_PERIOD/2 - 1
ticks, the function will "consider" a to actually predate b.

dhylands, 2016-10-31:
time_diff will return valid results for all possible inputs. If you happen to call time_diff with values which are are more than MICROPY_PY_UTIME_TICKS_PERIOD/2 - 1 apart then you might say that it produces incorrect results, but there is always a scenario where the exact same inputs are considered valid and less than MICROPY_PY_UTIME_TICKS_PERIOD/2 - 1 ticks apart, and there is NO way for time_diff to know which was intended.

pfalcon, 2016-10-31:
>> Well, it is possible and when it happens it's dealt with in a specific way, as the commit message says:

> if a follows b in more than MICROPY_PY_UTIME_TICKS_PERIOD/2 - 1 ticks, the function will "consider" a to actually predate b.

Yes, and yet it's pretty mundane situation every human knows how to deal with. In the latest commit 
https://github.com/micropython/micropython/commit/805f7ea2f2b4e1aeef7326db439ab2b810729f33 (which introduces docs for ticks_add() btw) I added such informal diversion:

>   Informal rationale of the constraints above: Suppose you are locked in a room with no
   means to monitor passing of time except a standard 12-notch clock. Then if you look at
   dial-plate now, and don't look again for another 13 hours (e.g., if you fall for a
   long sleep), then once you finally look again, it may seem to you that only 1 hour
   has passed. To avoid this mistake, just look at the clock regularly. Your application
   should do the same. "Too long sleep" metaphor also maps directly to application
   behavior: don't let your application run any single task for too long. Run tasks
   in steps, and do time-keeping inbetween.

pfalcon, 2016-11-05:
This is fully finished now. https://github.com/micropython/micropython/issues/2219 is next in queue.

closed 2016-11-05 by pfalcon