micropython/micropython #2218: Please remove const_table member from mp_code_state structure

pfalcon, 2016-07-03:
The idea of mp_code_state structure (note: it lacks _t suffix, worth fixing too) is to contain fields representing volatile code execution context. It should not contain any static fields, inferrable from bytecode. However, 713ea1800d1f0c82a0c75885ad034705556ab5ef added const_table pointer to each, which is well, constant for a particular bytecode.

Extending the idea further, even the original n_state member is superfluous, as it's static for bytecode. It's used to setup needed pointers in mp_execute_bytecode(), and that's where it should be decoded from mp_code_state::code_info.

That will cut size of mp_code_state, as used e.g. for generator state to 5 mandatory words and will allow (many) more generators to fit in 2 gc blocks.

dpgeorge, 2016-08-10:
The const_table pointer cannot be obtained from the bytecode pointer.  It's a separate chunk of memory.

Actually, if you don't use persistent bytecode then const_table is not needed in the mp_code_state structure, so that's something that can be fixed (but won't affect esp8266, stmhal or unix ports).

As for n_state, it could be computed, but you'd need to store a pointer to the very start of bytecode in mp_code_state (perhaps replacing code_info, and then you'd need to also compute code_info).

All these modifications would reduce performance of the VM because of the extra decoding.  They would also increase stack size if you cached the values in local variables in the VM.

pfalcon, 2016-08-11:
Well, the point of this report is that mp_code_state, in case of a generator, should contain only state of a generator, plus a pointer to a generator function info from which it was created. Constant information about generator function should not be duplicated again and again in every mp_code_state, wasting memory. Unless there were testing to back such optimization, in which case it should be documented in code comments. (And it wouldn't give 2x performance increase, and I don't think that even 5% is worth it, the situation we have is constant lack of memory, and that what should be optimized for).

P.S. do we want to rename mp_code_state to mp_code_state_t?

dpgeorge, 2016-08-11:
I have a few ideas how to reduce the size of mp_code_state, but they require some deep changes (especially to the native code emitter, which also uses mp_code_state and doesn't have the concept of bytecode) and also require lots of profiling to check stack usage and performance regression.

> P.S. do we want to rename mp_code_state to mp_code_state_t?

Yes.

pfalcon, 2016-08-11:
Ok, this is not urgent. It was created to not forget about it, and I assigned it to you recently just a kind of triaging (with all the recent influx of non-bugreprots, it's hard to spot to work on). Feel free to unassign if it helps to better organize it on your side.

dpgeorge, 2017-03-17:
I ended up coming at this from a totally different angle, see #2963.  A proper fix for that bug lead to removal of const_table, as well as code_info and n_state from the structure.  They were replaced with a pointer to the underyling function object (from which one can obtain all this other read-only data).

Measuring stack usage, the fix actually allowed to decrease stack use by 8-16 bytes per Python function call, which is also a big win.  Heap usage will be reduced (for generators) and performance of the VM was measured to be about the same.

See 71a3d6ec3bd02c5bd13334537e1bd146bb643bad for the patch.

closed 2017-03-17 by dpgeorge