micropython/micropython #2219: RFC: Optimized heap queue implementation for uasyncio

pfalcon, 2016-07-03:
Switch uasyncio to wrap-around time source like utime.ticks_ms(), as required for baremetal ports, requires changing to the comparison done in uheapq module - it should use partial ordering definable for modular arithmetic (ring math object). So, that requires changing uheapq anyway, and I currently have adhoc changes for that coded.

Besides this must-have feature, there're also change to optimize:

1. Pre-allocate queue of some size.
2. Avoid constant packing and unpacking of tuples as done now.

So, that actually calls for introduction of adhoc heap queue class implementation for uasyncio. The main question is how to implement it - I consider turning current moduheapq.c into header with poor-man's templating implementation to teach it working with different data structures. Any objections?

After this changes, there will be (in a common case) the only remaining source of memory traffic for uasyncio usage - constant generator creation from yield from.

pfalcon, 2016-07-03:
>  and I currently have adhoc changes for that coded.

For reference: https://github.com/pfalcon/micropython/commit/75de07a86badbd60b373bd2d965a05202730dca0

pfalcon, 2016-11-10:
@dpgeorge : So, considering current situation and backlog, I'd rather proceed with step by step changes, and merge change in the previous comment largely as is (after required clean up of course). This will allow to switch unix implementation and test it well, and uasyncio.core (I/O-less) can be used on esp8266 for testing too. Let me know if there're any objects.

dpgeorge, 2016-11-11:
> I'd rather proceed with step by step changes, and merge change in the previous comment largely as is (after required clean up of course).

From inspection of that patch it seems that it adds extra args to existing heapq methods, and so retains backwards compatibility.  If so then +1 to merge.

pfalcon, 2016-11-11:
@dpgeorge : Yes, sure, backwards compatibility is retained. The only concern is that later it may need to changed again, then again, and again. But as it involves many moving parts, we'd better really do it like that, step by step, without trying to arrive at a final destination right away. So, some churn would be expected.

pfalcon, 2016-11-11:
Refactored and merged as https://github.com/micropython/micropython/commit/0cbc07227c4ab70c846f40a1e2a5fd57dec30428.

closed 2016-11-11 by pfalcon