micropython/micropython #2220: Support WiThumb / MPU6050 ?

petri, 2016-07-04:
There's another [ESP8266-based campaign](https://www.kickstarter.com/projects/hackarobot/withumb-arduino-compatible-wifi-usb-thumb-imu) that's just been funded.

It's basically a USB stick with WiFi and some sensors:

- temperature using MCP9808
- gyro + accelerometer using [MPU6050](https://www.invensense.com/products/motion-tracking/6-axis/mpu-6050/)

The [micropython on ESP8266 campaign](https://www.kickstarter.com/projects/214379695/micropython-on-the-esp8266-beautifully-easy-iot/description) has promised to provide support for the MCP9808 so I assume that part already will work?

It would be fabulous if the WiThumb were fully programmable with micropython as well. But that would require support for the MPU6050 as well.

turbinenreiter, 2016-07-04:
https://github.com/micropython-IMU/micropython-mpu9x50

The base class  `imu` is the driver for the MPU6050.

pfalcon, 2016-07-05:
If you would like WiThumb to support MicroPython, please let *them* know that. From our side, we already support a generic ESP8266 port with tons of features. But supporting a particular product first of all lies on its vendor.

closed 2016-07-05 by pfalcon

petri, 2016-12-10:
In case someone finds this issue looking for information, MicroPython installs smoothly on WiThumb:

https://koodaamo.wordpress.com/2016/12/10/micropython-on-the-withumb-on-osx/