micropython/micropython #2223: unix: build fails when USSL is enabled

dhylands, 2016-07-06:
If I check out the latest tree, update submodules and do a clean build (in the unix tree), then I get the following build error (DEBUG or non-DEBUG):
```
CC ../py/../extmod/modussl.c
In file included from ../lib/axtls/ssl/tls1.h:47:0,
                 from ../lib/axtls/ssl/ssl.h:75,
                 from ../py/../extmod/modussl.c:37:
../lib/axtls/ssl/crypto_misc.h:88:15: error: flexible array member in otherwise empty struct
     X509_CTX *cert[CONFIG_X509_MAX_CA_CERTS];
               ^
make: *** [build/py/../extmod/modussl.o] Error 1
```
This is under ubuntu 14.04 using gcc 4.8.4:
```
537 >gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I can work around this by using: ```make MICROPY_PY_USSL=0```

It looks like CONFIG_X509_MAX_CA_CERTS is defined to be nothing. (axtls/config/config.h)

pfalcon, 2016-07-07:
Why the official CI build is green then?

dhylands, 2016-07-07:
> Why the official CI build is green then?

Perhaps it uses a different version of gcc?

If we don't support 4.8 then we should probably say that someplace.

pfalcon, 2016-07-11:
> Perhaps it uses a different version of gcc?

Let's see: https://travis-ci.org/micropython/micropython/builds/143987589#L189 .

I'm a bit shy to say a fellow developer that if the CI is green and his build is red, he should be looking at what's wrong with *his* system. And well, it actually would be fun to find a CI to be broken in such a way that it's green while it should have been red. (Of course, the build is green locally for me too, otherwise I would rush to fix it.)

dhylands, 2016-07-11:
I figured out that I forgot to do the 'make axtls' (so it was using some combination of new and old axtls files) and then things started working again.

Closing.

closed 2016-07-11 by dhylands