micropython/micropython #2224: Large Tests Fail on Windows

ryannathans, 2016-07-07:
Attempting to run the following with a pyboard
`py -3 run-tests --device COM63 --target pyboard`

on Linux all tests pass
on Windows the large tests fail. A perfect example of this is the test basics/int_big_div.py.

The generated output from windows python is perfect, but the data obtained from the pyboard appears to be corrupted in random locations when compared. This corruption does not occur on unix.

here is an example of the corruption:

`807129636248249042473169689192672273375203` from the device instead of 
`80712963624824904247523071397869999158397`

Sometimes the number is very small (eg 5 digits) when it should be much larger, sometimes the error manifests as extra new lines (0 length digits), sometimes the numbers outputted are very large (eg: 20+ digits longer than they should be). To clarify, random sections of the output from the device appear to be corrupted, these sections change every time the test is run. Oddly, these corruptions do not introduce unusual characters/symbols, just adding or removing numbers.

This behavior does not occur when testing with linux.

ryannathans, 2016-08-08:
Interestingly, just got tests to work on a Windows 8 tablet.

Seems to be a problem with the **combination of Windows and particular USB chipsets**, linux on the same "faulty" pc works without issue.

dpgeorge, 2016-08-08:
I did see very similar behaviour under Linux, but it's very hard to reproduce.  It could be a bug in the pyboard VCP implementation where characters in the outgoing buffer are lost or jumbled up.

ryannathans, 2016-09-22:
Work around: connect USB to serial adapter to a UART on the device and in boot.py set pyb.repl_uart() to the uart of your adaptor. Testing seems to work without going through USB, yay

dpgeorge, 2017-06-14:
This was fixed by 9a8e7f7a8e93e9319d2ff5795f4b8a6b9e4ba5ad

closed 2017-06-14 by dpgeorge