micropython/micropython #2225: Build error for CC3200 application

chenek, 2016-07-07:
I tried to build micropython for CC3200 in Cygwin with "make BTARGET=application BTYPE=release BOARD=LAUNCHXL". However, I always get the following error:

$ make BTARGET=application BTYPE=release BOARD=LAUNCHXL
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Create build/LAUNCHXL/release/genhdr/pins.h
/c/cygwin64/bin/python: /c/cygwin64/bin/python: cannot execute binary file
application.mk:248: recipe for target 'build/LAUNCHXL/release/genhdr/pins.h' failed
make: *** [build/LAUNCHXL/release/genhdr/pins.h] Error 126

How can I fix this?

dhylands, 2016-07-07:
This seems to be the relevant error:  /c/cygwin64/bin/python: cannot execute binary file

Are you able to execute python directly?
Are you on a 64 bit machine?

chenek, 2016-07-07:
Yes, I can execute python directly and I run Cygwin on a 64 bit machine.

dhylands, 2016-07-07:
More specifically can you run '/c/cygwin64/bin/python'?
Are you using a Win32 version of make? or the cygwin version of make?

chenek, 2016-07-08:
Yes, I can run "/c/cygwin/bin/python". I am using Cygwin version of make.

dhylands, 2016-07-08:
The path in question was /c/cygwin64/bin/python

chenek, 2016-07-08:
Yes, I can run /c/Cygwin64/bin/python . It's typo in my previous reply.

dhylands, 2016-07-08:
So I'm baffled why make can't execute python while you can execute it from the command line. The fact that it thinks its a binary file usually boils down to permissions or some type of wrong architecture for the file, or perhaps using a 32-bit make trying to execute 64-bit python?

I've got a similar setup (64-bit cygwin) and everything seems to be working fine for me.

pfalcon, 2016-07-30:
Appears to be not reproducible and thus likely an error on user side. @chenek , you may want to re-install cygwin from scratch as one way to diagnose it further. Feel free to reopen when exact steps to reproduce it will be found.

closed 2016-07-30 by pfalcon