micropython/micropython #2228: ARM debug build fails

tralamazza, 2016-07-08:
#### How to reproduce (*cc3200*)
- Edit [application.mk:197](https://github.com/micropython/micropython/blob/master/cc3200/application.mk#L197) to `CFLAGS += -DDEBUG` (`NDEBUG` looks like a bug)
- `make BTARGET=application BTYPE=debug BOARD=LAUNCHXL`

```sh
CC ../py/mpstate.c
CC ../py/nlrthumb.c
In file included from ../py/nlr.h:34:0,
                 from ../py/mpstate.h:34,
                 from ../py/nlrthumb.c:27:
../py/mpstate.h:223:50: error: expected ')' before '->' token
 #define MP_STATE_THREAD(x) (mp_thread_get_state()->x)
                                                  ^
../py/nlr.h:104:12: note: in expansion of macro 'MP_STATE_THREAD'
     assert(MP_STATE_THREAD(nlr_top) != val),nlr_push(val)
            ^~~~~~~~~~~~~~~
../py/nlrthumb.c:38:37: note: in expansion of macro 'nlr_push'
 __attribute__((naked)) unsigned int nlr_push(nlr_buf_t *nlr) {
                                     ^~~~~~~~
../py/nlr.h:104:37: error: expected ')' before '!=' token
     assert(MP_STATE_THREAD(nlr_top) != val),nlr_push(val)
                                     ^
../py/nlrthumb.c:38:37: note: in expansion of macro 'nlr_push'
 __attribute__((naked)) unsigned int nlr_push(nlr_buf_t *nlr) {
                                     ^~~~~~~~
../py/nlr.h:104:5: error: expected ')' before '?' token
     assert(MP_STATE_THREAD(nlr_top) != val),nlr_push(val)
     ^
../py/nlrthumb.c:38:37: note: in expansion of macro 'nlr_push'
 __attribute__((naked)) unsigned int nlr_push(nlr_buf_t *nlr) {
                                     ^~~~~~~~
make: *** [../py/mkrules.mk:47: build/LAUNCHXL/debug/py/nlrthumb.o] Error 1
```

#### How to reproduce (*minimal*)
- `make CROSS=1 DEBUG=1`

```
CC ../py/mpstate.c
CC ../py/nlrthumb.c
In file included from ../py/nlr.h:34:0,
                 from ../py/mpstate.h:34,
                 from ../py/nlrthumb.c:27:
../py/mpstate.h:225:41: error: expected ')' before '.' token
 #define MP_STATE_THREAD(x) (mp_state_ctx.thread.x)
                                         ^
../py/nlr.h:104:12: note: in expansion of macro 'MP_STATE_THREAD'
     assert(MP_STATE_THREAD(nlr_top) != &val),nlr_push(val)
            ^~~~~~~~~~~~~~~
../py/nlrthumb.c:38:37: note: in expansion of macro 'nlr_push'
 __attribute__((naked)) unsigned int nlr_push(nlr_buf_t *nlr) {
                                     ^~~~~~~~
../py/nlr.h:104:37: error: expected ')' before '!=' token
     assert(MP_STATE_THREAD(nlr_top) != &val),nlr_push(val)
                                     ^
../py/nlrthumb.c:38:37: note: in expansion of macro 'nlr_push'
 __attribute__((naked)) unsigned int nlr_push(nlr_buf_t *nlr) {
                                     ^~~~~~~~
../py/nlr.h:104:5: error: expected ')' before '?' token
     assert(MP_STATE_THREAD(nlr_top) != &val),nlr_push(val)
     ^
../py/nlrthumb.c:38:37: note: in expansion of macro 'nlr_push'
 __attribute__((naked)) unsigned int nlr_push(nlr_buf_t *nlr) {
                                     ^~~~~~~~
make: *** [../py/mkrules.mk:47: build/py/nlrthumb.o] Error 1
```

tralamazza, 2016-07-10:
For now I'm just removing the assert in https://github.com/micropython/micropython/blob/master/py/nlr.h#L104 and just calling `nlr_push(val)`

tralamazza, 2016-07-11:
The `-DNDEBUG` for debug builds look weird, can anyone confirm that's the intended behavior?

pfalcon, 2016-07-11:
Cannot reproduce:

~~~
micropython/minimal$ make CROSS=1 DEBUG=1
LINK build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 150868	      4	   2524	 153396	  25734	build/firmware.elf
Create build/firmware.dfu
~~~

~~~
$ arm-none-eabi-gcc -v
...
gcc version 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977] (GNU Tools for ARM Embedded Processors) 
~~~

tralamazza, 2016-07-11:
Sorry @pfalcon I  forgot to add my Makefile edit (Makefile:32 `-DDEBUG`).

tralamazza, 2016-07-13:
@pfalcon are you able to reproduce now?

pfalcon, 2016-07-13:
I actually don't completely understand what's the topic of this bugreport. Let's start with basics: in MicroPython, you produce a debug build with `make DEBUG=1`. Period.

Going anywhere from that, there should be clear answers to questions: a) what do  you want to achieve? b) why do you achieve it in a way you do?

thomasfrimannkoren, 2016-07-13:
I had the same problem that @tralamazza  experienced, and I believe it is a bug. 

It is caused by the expansion of the macro from py/nlr.h:103 when compiling py/nlrthumb.c.
This expansion causes the function header of nlr_top in py/nlrthumb.c:38 to become

`__attribute__((naked)) unsigned int (((mp_state_ctx.thread.nlr_top) != nlr_buf_t *nlr) ? (void)0 : __assert_func ("C:/Users/tfkoren/Work/t173_tbsense_micropython/mod/micropython/py/nlrthumb.c", 38, __func__, "MP_STATE_THREAD(nlr_top) != nlr_buf_t *nlr")),nlr_push(nlr_buf_t *nlr) {
`

When debug is turned off the assertion dissappears, and so does the problem.

tralamazza, 2016-07-13:
[@pfalcon](url) I've updated the topic name, maybe that helps?

> a) what do you want to achieve?

I want to use the `DEBUG` define as I've seen it used throughout the code (also because it is a standard). A quick search for `#ifdef DEBUG` reveals that.

> b) why do you achieve it in a way you do?

I don't understand this question.

I've reported 2 ways on how to reproduce this. Moreover, I think the CC3200 Makefile has a separate bug when it defines `NDEBUG` in both debug and release builds. Since I haven't found this documented anywhere I made this bug report.

dhylands, 2016-07-13:
The -NDEBUG is turning off assertions (on the cc3200 build)

It's quite possible that a DEBUG build would be too big with assertions enabled.

Looking at git blame, it appears that the change from -DDEBUG to -DNDEBUG was deliberate:
https://github.com/micropython/micropython/commit/f5248a087ac5667c53b7eab0e874b517ddf53dae

So you'd probably need to ask @danicampora for clarification.

pfalcon, 2016-07-20:
According to @dhylands , this should be fixed by https://github.com/micropython/micropython/pull/2241.

closed 2016-07-20 by pfalcon

tralamazza, 2016-07-21:
partially, the Makefiles are still the same.