micropython/micropython #2230: Add details about running files on startup to the documentation.

skorokithakis, 2016-07-10:
I read the ESP8266 documentation and it seems to leave a few questions open:

* How do I add a file to run whenever the ESP starts up?
* Can I send it over webrepl or save it in a console, or do i need to flash it in the firmware?
* How do I create modules? Does the standard Python way work?

I think I can probably guess the answers to a few of these, but it would be great if they were definitively spelled out in the docs.

deshipu, 2016-07-11:
The boot process is described at http://docs.micropython.org/en/latest/esp8266/esp8266/general.html#boot-process

The webrepl is described at http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html#webrepl-web-browser-interactive-prompt

Uploading files is covered at http://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/filesystem.html

skorokithakis, 2016-07-11:
I saw those (except the first one), but I think this is such a common use case that it deserves its own section. I tried copying scripts to the filesystem using the web REPL and the CLI script but neither worked for me, and the process felt unsupported.

It seems to me that almost everyone who wants to use a microcontroller wants to have it permanently running some code, and, while the REPL is fantastic for prototyping, I think the process of uploading code should be better documented and more streamlined.

deshipu, 2016-07-11:
You have to remember that this is still an experimental port, and a lot of things are being still worked on. For instance, since recently, you can upload (and download) files directly with your web browser, which should make things much easier. This is, however, new, so it will take some time to settle, be tested, and make it into the tutorials.

The documentation is also being created as things are added, and may be incomplete or fragmented. Please feel free to propose pull requests wherever you see a way to improve it, this kind of specific feedback is very valuable. (I'm not one of the core developers, but I'm sure they will gladly accept the pull requests if they improve the docs.)

Cabalist, 2016-07-11:
If you are compiling your own firmware this was the solution offered to me and I think the most useable at the moment.

https://forum.micropython.org/viewtopic.php?f=15&t=2114&e=1&view=unread#unread

Basically you just put the code you want executed into the scripts/ folder and edit main.c to add:
```
pyexec_frozen_module("myscript.py");
```

skorokithakis, 2016-07-11:
Ah, that looks interesting, thanks. Personally, I'm using the `webrepl_cli.py` script to upload/update my script to the ESP, which is great because it lets me both log in to the web REPL and remotely update scripts.

NTL2009, 2016-10-25:
Thread is kinda old I know, but I came across it in a search, so this might help others.

I'm not very fluent in all these tools, hadn't done any serial comm and AT commands for 30 years? I'm using the HiLetgo  NodeMCU LUA ESP8266 (1MB flash - USB connection). I was able to send the AT commands using:   cu -l /dev/ttyUSB0 -s 115200   
but was confused until I read you type the AT command, then ENTER and then CNTRL-J to send the command. Us newbies just don't know this stuff, and it is not spelled out. It took a lot of searching to find this essential info. Maybe there are better ways, but this worked.

After confirming AT commands got the proper response from the boards, I managed to get it reflashed with MicroPython using esptool pretty easily. **But I was also confused on how to copy my script over to the ESP module.** And I preferred to do it wired, over USB, as I already had that comm set up. The solution I found was an app called 'ampy' (there might be better alternatives, but this worked for me). 

To install 'ampy' on my 14.04 Xubuntu machine:   ~$ sudo pip2 install adafruit-ampy
To copy (put) my 'main.py' script from my HOME dir to the module:  ~$ ampy -p /dev/ttyUSB0 put main.py

I reset the board, and then using:  ~$ picocom /dev/ttyUSB0 -b 115200 for communication, the little counting program I wrote worked as expected.

Hopefully, these little tips will help fill in the blanks for the less experienced users. I could probably find some time to help with the documentation, if that makes sense.

pfalcon, 2017-05-13:
Assuming that links above have clarified the situation. If there're still issues, please open separate tickets, with exact steps to reproduce. First step should be "1. Have read all the documentation at (link), but ...". Thanks.

closed 2017-05-13 by pfalcon