micropython/micropython #2231: Include mock pyb module for non-MCU testing

nmz787, 2016-07-11:
It seems rshell is hanging when I `cp` buggy files, but I can't get a sense of my bugs because I can't pre-test on my laptop and find syntax errors for example, non-runtime errors at least. If there was a mock pyb module, then imports wouldn't fail.

dhylands, 2016-07-11:
rshell isn't part of micropython, so this isn't the best place to discuss this.

If the buggy files are part of boot.py or main.py then that will definitely mess with rshell.

You can create your own mock pyb. I created one for some other stuff I was working on here:
https://github.com/dhylands/bioloid3/blob/master/pyb.py
I plan on fleshing that out as I need more functionality mocked.

nmz787, 2016-07-11:
Does rshell really matter when it comes to bench-testing though?

dhylands, 2016-07-11:
> Does rshell really matter when it comes to bench-testing though?

I guess I don't understand your comment. You'll need to define "bench testing". And this isn't the place to be discussing rshell issues (this is for micropython issues).

nmz787, 2016-07-11:
I removed all references to rshell, since I am unsure it is related or not, and I really want this ticket to be about ensuring my code is syntactically error before moving it from my personal computer to the micropython MCU.

coylen, 2016-07-12:
If I correcty understand what you are asking for then I think you will find the repository listed below as meeting your needs.

https://github.com/dastultz/micropython_pyb

nmz787, 2016-07-12:
> https://github.com/dastultz/micropython_pyb
I am not home to test this, but it looks to be what I am after... all the functions and classes with 'stub' or 'skeleton' implementations... just enough so imports won't fail, so this way the interpreter can find syntax issues (and as the repo lists,  will also help autocomplete if you're in an IDE).

dhylands has a much smaller version of this mock module.... so there are (at least) two folks already coming up with these kind of solutions... so seems natural that this mock-module be part of micropython-proper... since that way the actual pyb and the mock-pyb will never be out-of-sync.

Will test soon, thanks!

dhylands, 2016-07-12:
In all fairness, mine only covered ADC and Pin classes, and then only implemented what I actually called.

nmz787, 2018-05-06:
bump... seems like this should be easy to do for someone who knows the micropython directory structure... just grep for all the `def` statements in the relevant source files, then for each `def` found, create an empty mock function with args `*args, **kwargs`, but no body (or a `pass` statement).

nmz787, 2018-05-06:
@coylen and @dastultz I had to add a few things and change some strange lone `*` characters in some function/method args to `*args, **kwargs` instead... but for my current debug session these helped me find some non-ascii characters that crept into my code somehow, which apparently MicroPython isn't raising a good exception, as @jgriessen couldn't detect them and in-situ (i.e. uploading the Python files to his board and importing)
micropython mock library - for checking syntax of your MicroPython from Linux
https://gist.github.com/nmz787/771774b5ceae4cedf257e336e3006352

If this was automated per-release, with matching function/method signatures (instead of quick and dirty `*args, **kwargs`), then that would be even better for syntax coverage of user scripts.

dastultz, 2018-05-07:
@nmz787 I've applied your changes to my repo.

peterhinch, 2018-05-07:
The "strange lone `*` characters" are Python syntax for declaring that subsequent arguments are keyword-only args. See [pep 3102](https://www.python.org/dev/peps/pep-3102/).

nmz787, 2018-05-07:
Ah, that's Python3-only syntax, didn't know about that PEP before.

jgriessen, 2018-05-07:
I got good results testing with this new update.

projectgus, 2024-07-05:
Closing as this seems to be long resolved now.

closed 2024-07-05 by projectgus