micropython/micropython #2232: Update LED intensity related documentation to be useful

nmz787, 2016-07-11:
The documentation here:
https://docs.micropython.org/en/latest/pyboard/pyboard/tutorial/leds.html?highlight=led%20intensity#the-fourth-special-led

is misleading and does not work when used. I asked on the forum about this, and dhyland's reply was:

> LED.intensity will only work if there is a Timer PWM output pin connected to the LED and things have been configured properly.

Which is clearly missing from the example snippet.

dhylands, 2016-07-11:
The example snippet is for the pyboard, and is correct (and the documentation you quoted is for the pyboard).

If you're using a different board, then there isn't any guarantee that the pyboard examples will work.

nmz787, 2016-07-11:
When I go to [https://docs.micropython.org](https://docs.micropython.org) I get redirected to a URL with 'pyboard' in it, but I just assumed that was long-hand for the 'pyb' module... and thus is documentation that would be common for anything implementing Micropython. Even the class-doc here (which is also under the pyboard URL hierarchy) mentions nothing about using the Timer class and the setup required:
https://docs.micropython.org/en/latest/pyboard/library/pyb.LED.html?highlight=led#pyb.LED.intensity

nmz787, 2016-07-11:
I guess I see now "MicroPython runs on a variety of systems and each has their own specific documentation. You are currently viewing the documentation for the pyboard."
but don't see any docs for anything but pyboard on that website.

deshipu, 2016-07-11:
In the bottom-left corner of the screen you have "Ports and versions" menu, where you can select for which board you want the documentation.

pfalcon, 2016-07-17:
Hopefully the above responses clarify how to work with http://docs.micropython.org/ site.

closed 2016-07-17 by pfalcon

nmz787, 2016-07-17:
It isn't helpful for beginners, I still think the documentation could be
more precise, complete, and clear.

pfalcon, 2016-07-17:
That's true for any documentation out there. Good thing about open-source documentation is that everyone can help to improve it. So, if you have specific improvements in mind, feel free to submit patches for them (the standard recommendation is to start with simple and short changes).

nmz787, 2016-07-18:
Yes but the whole point of this ticket, is to have a flag to remind people 'in the know' that beginners are having trouble with the docs. Closing it doesn't magically fix this issue for beginners, it is just brushing the issue aside because you don't seem to want to acknowledge or fix the issue yourself. I don't know the infrastructure of the project well enough to go fix it, but I am smart enough to know to file a ticket when I had obvious struggles. I would like to see this ticket re-opened until _something_ with the documentation changes that actually solves this issue.

dpgeorge, 2016-08-29:
> In the bottom-left corner of the screen you have "Ports and versions" menu, where you can select for which board you want the documentation.

For people who know about this menu it's very useful.  But for someone coming to the project for the first time it's not at all obvious how to use the docs (eg there's nothing on the front title page telling you that you can change "ports").

Because the project is getting more sophisticated, with more boards being supported in the docs, I think we need a "docs for the docs".  It can be very simple, taking the following form: the front page of the docs (at the top-level url http://docs.micropython.org) can be a welcome page with a short description about how to navigate the docs (and disclaimers like that at the top of http://docs.micropython.org/en/latest/pyboard/library/index.html), and then some big buttons (with images, like on the forum) that take you to the docs for each individual port (including microbit, which would link to the independent http://microbit-micropython.readthedocs.io/en/latest/).

@nmz787 would that have helped your case?

nmz787, 2016-09-06:
A better welcome/landing page probably would NOT have helped in this case, as the documentation doesn't cover platforms other than "wipy esp8266 unix pyboard'... which didn't cover any of the cases I was using (first an STM Nucleo dev board, then some other similar board that dhylands recommended for better USB, which was sold as a .NET dev board). It seems the real answer is that there is not non-platform-specific documentation. If there was a non-platform-specific docs area, then my first comment about making sure to tie a PWM timer to a pin for LED.intensity to work would hold true, even for pyboard, etc which might have helper code baked in to their calls.