micropython/micropython #2233: macOS: mp_thread_get_state undefined symbol

ilovezfs, 2016-07-11:
version 1.8.2
```
LINK micropython
Undefined symbols for architecture x86_64:
  "mp_thread_get_state", referenced from:
      _nlr_push in nlrx64.o
      _nlr_pop in nlrx64.o
      _nlr_jump in nlrx64.o
     (maybe you meant: _mp_thread_get_state)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [micropython] Error 1
```

ilovezfs, 2016-07-11:
Setting MICROPY_PY_THREAD = 0 with string replacements fixes it but it would be great if it could be fixed properly, or at least MICROPY_PY_THREAD set to 0 for macOS until it's figured out.

dpgeorge, 2016-07-11:
Should be fixed by 3593d8e10c960fdc9349589deec3a2d8e4658cf3, please test.

ilovezfs, 2016-07-11:
@dpgeorge Yup. Nice work. Any chance of getting a new tag to simplify https://github.com/Homebrew/homebrew-core/pull/2607? :innocent:

closed 2016-07-11 by ilovezfs

dpgeorge, 2016-07-11:
Great!  Are you able to verify that the thread tests actually work: `cd tests && ./run-tests -d thread`?

> Any chance of getting a new tag

Not at this point since the build is a bit unstable at the moment.  Please just use the git hash.

ilovezfs, 2016-07-11:
> Please just use the git hash.

We can't define a version by a specific hash unless it's a tag (since we'd just be inventing the version number that goes with that hash).

But I could apply the individual commit as a patch if necessary. Are there other commits you think should be applied on top of the 1.8.2 tag?

ilovezfs, 2016-07-11:
Thread tests:
```
bash-3.2$ cd tests && ./run-tests -d thread
skip  thread/mutate_bytearray.py
skip  thread/mutate_dict.py
skip  thread/mutate_instance.py
skip  thread/mutate_list.py
skip  thread/mutate_set.py
pass  thread/stress_aes.py
pass  thread/stress_heap.py
pass  thread/stress_recurse.py
pass  thread/thread_exc1.py
pass  thread/thread_exit1.py
pass  thread/thread_exit2.py
pass  thread/thread_gc1.py
pass  thread/thread_ident1.py
pass  thread/thread_lock1.py
pass  thread/thread_lock2.py
pass  thread/thread_lock3.py
pass  thread/thread_lock4.py
pass  thread/thread_qstr1.py
pass  thread/thread_shared1.py
pass  thread/thread_shared2.py
pass  thread/thread_sleep1.py
pass  thread/thread_stacksize1.py
pass  thread/thread_start1.py
pass  thread/thread_start2.py
19 tests performed (123 individual testcases)
19 tests passed
5 tests skipped: mutate_bytearray mutate_dict mutate_instance mutate_list mutate_set
```
All tests:
https://gist.github.com/ilovezfs/03429568daa9c4d82184488b2961980d

pfalcon, 2016-07-11:
> Not at this point since the build is a bit unstable at the moment. Please just use the git hash.

@dpgeorge , if you noticed, there were some random glitches with threading tests in travis.

dpgeorge, 2016-07-11:
> if you noticed, there were some random glitches with threading tests in travis.

Ok, I see them.  I just made a few changes to try and fix it, but I can't reproduce those particular errors in stress_heap and thread_lock4.