micropython/micropython #2236: esp8266 Pin 16 works only as an output

peterhinch, 2016-07-12:
Tested on an Adafruit Feather Huzzah, with firmware built from source 11th July. As an input the pin reads 1 even when connected to 0V. It works as an output. If this behaviour is by design the docs should reflect it.
```
[adminpete@axolotl]: ~
$ mtusb
--- Miniterm on /dev/ttyUSB0: 115200,8,N,1 ---
--- Quit: Ctrl+]  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

>>> 
connected with misspiggy, channel 3
dhcp client start...
ip:192.168.0.23,mask:255.255.255.0,gw:192.168.0.1

>>> from machine import Pin
>>> p16 = Pin(16, Pin.IN, Pin.PULL_UP)
>>> p16()
1
>>> p16() [COMMENT: wired to 0V]
1
>>> p16 = Pin(16, Pin.OUT) [COMMENT: wired to LED connected to 0V via resistor]
>>> p16(0) [LED is off]
>>> p16(1) [LED is on]
>>> 
```

pfalcon, 2016-08-09:
@dpgeorge : So, when you get back to generic esp8266 stuff, this one is probably worth being the first to be looked into.

dpgeorge, 2016-08-10:
Should be fixed by b203c1774e4eb6bb6bc04b99d0a5f06f0aa04e40.  @peterhinch if you could please confirm it works and then close the issue, thanks!  Also note that PULL_UP is not supported for Pin(16).

peterhinch, 2016-08-10:
Pin now works as input and output :)

closed 2016-08-10 by peterhinch