micropython/micropython #2237: 1.8.2 ESP8266 Issues setting RTC

Cabalist, 2016-07-12:
I am getting some odd behaviour when I try and set the RTC using a time tuple.   Code to reproduce:

```
>>> import machine
>>> import time
>>> time.localtime()
(2000, 1, 1, 0, 0, 17, 5, 1)
>>> rtc = machine.RTC()
>>> rtc.datetime() # 5 hour difference from the time module
(2000, 1, 1, 5, 0, 0, 32, 77)
>>> import ntptime
>>> x = ntptime.time()
>>> x
521665069
>>> time.localtime(x) # This is correct
(2016, 7, 12, 18, 57, 49, 1, 194)
>>> rtc.datetime(time.localtime(x)) # Set RTC with timetuple
>>> rtc.datetime() # Roughly 1 day 5 hours off.
(2016, 7, 14, 3, 9, 49, 11, 925)
```

Any thoughts?  I am building from the source tagged 1.8.2

Cabalist, 2016-07-12:
More weirdness simpler:

```
>>> from machine import RTC
>>> rtc = RTC()
>>> rtc.datetime((2016, 7, 12, 0, 0, 0, 0, 0))
>>> rtc.datetime()
(2016, 7, 12, 1, 0, 0, 1, 467)
```

It gains an hour.

Cabalist, 2016-07-12:
OK.  SO.  I should have been using ntptime.settime()

```python
def settime():
    t = time()
    import machine
    import utime
    tm = utime.localtime(t)
    tm = tm[0:3] + (0,) + tm[3:6] + (0,)
    machine.RTC().datetime(tm)
    print(utime.localtime())
```
Apparently the tuple that ntptime returns needs some reorganization to be fed into the RTC.   This gives me the results that I expect. 

This contradicts the tests that are here:
micropython/tests/pyb/rtc.py

I am going to leave this open because I would love someone to weigh in on this.   Seems like an odd API mismatch.  But my problem is solved.  Feel free to close.  :)

robert-hh, 2016-07-13:
The tuples returned by time.localtime() and rtc.datetime() are different (see documentation). If you feed the tuple returned by localtime into rtc.datetime, the result is wrong.
localtime(): (year, month, day, hours, minutes, seconds,  weekday, yearday)
datetime(): (year, month, day, weekday, hours, minutes, seconds, subseconds)
Like you, I am using ntptime.settime().

martyvis, 2016-08-31:
I believe the day of week has an off by one error. Setting for 31 August 2016, which is a Wednesday, and hence day of week should be 3. Either setting the RTC with 0 in the 4th field (so letting the function compute the day or week), or setting it the correct value of 3. (Based on the pyb documentation http://docs.micropython.org/en/v1.8.2/pyboard/library/pyb.RTC.html - weekday is 1-7 for Monday through Sunday.)

>>> rtc.datetime((2016,8,31,0,12,3,55,0))
>>> rtc.datetime()
(2016, 8, 31, **2**, 12, 3, 57, 197)
>>> rtc.datetime((2016,8,31,3,12,3,55,0))
>>> rtc.datetime()
(2016, 8, 31, **2**, 12, 3, 58, 597)
>>>

dpgeorge, 2016-09-02:
> I believe the day of week has an off by one error.

Thing are, unfortunately, inconsistent between the ports: pyboard's RTC uses 1-7 for the weekdays (following the hardware), and everything else uses 0-6 (following Python).  So the esp8266 doesn't have an off-by-one error.

eflukx, 2018-11-18:
Excuse me for necroposting this, but the `weekday` field  mentioned by @robert-hh in 

> datetime(): (year, month, day, weekday, hours, minutes, seconds, subseconds)

seems to be missing in the [docs]:(https://docs.micropython.org/en/latest/library/machine.RTC.html#machine.RTC.init)

dpgeorge, 2022-06-25:
The problem here was the inconsistency of RTC datetime tuples across ports.  But there's no actual bug.

closed 2022-06-25 by dpgeorge